mobizt/Firebase-ESP32 #3: Help me please

cruizg, 2019-02-04:
I have a question With Wifi Connection. Do They have Support?.

cruizg, 2019-02-04:
Hi, i Use this post:
https://desire.giesecke.tk/index.php/2018/04/06/esp32-wifi-setup-over-ble/
merge with your library, but the message is connection refused, can you help me please:

`#include <BLEServer.h>
#include <BLEDevice.h>
#include <BLEAdvertising.h>
#include <Preferences.h>
//#include <IOXhop_FirebaseESP32.h>
#include "FirebaseESP32.h"
FirebaseData firebaseData;

#define FIREBASE_HOST "<URL_FIREBASE_Project>"
#define FIREBASE_AUTH "<PASSWORD>"
#include <SPI.h>

/** Build time */
const char compileDate[] = __DATE__ " " __TIME__;

/** Unique device name */
char apName[] = "KNOCK-xxxxxxxxxxxx";
/** Selected network 
    true = use primary network
    false = use secondary network
*/
bool usePrimAP = true;
/** Flag if stored AP credentials are available */
bool hasCredentials = false;
/** Connection status */
volatile bool isConnected = false;
/** Connection change status */
bool connStatusChanged = false;

/**
 * Create unique device name from MAC address
 **/
void createName() {
  uint8_t baseMac[6];
  // Get MAC address for WiFi station
  esp_read_mac(baseMac, ESP_MAC_WIFI_STA);
  // Write unique name into apName
  sprintf(apName, "KNOCK-%02X%02X%02X%02X%02X%02X", baseMac[0], baseMac[1], baseMac[2], baseMac[3], baseMac[4], baseMac[5]);
}

// List of Service and Characteristic UUIDs
#define SERVICE_UUID  "0000aaaa-ead2-11e7-80c1-9a214cf093ae"
#define WIFI_UUID     "00005555-ead2-11e7-80c1-9a214cf093ae"

/** SSIDs of local WiFi networks */
String ssidPrim;
String ssidSec;
/** Password for local WiFi network */
String pwPrim;
String pwSec;

/** Characteristic for digital output */
BLECharacteristic *pCharacteristicWiFi;
/** BLE Advertiser */
BLEAdvertising* pAdvertising;
/** BLE Service */
BLEService *pService;
/** BLE Server */
BLEServer *pServer;

/** Buffer for JSON string */
// MAx size is 51 bytes for frame: 
// {"ssidPrim":"","pwPrim":"","ssidSec":"","pwSec":""}
// + 4 x 32 bytes for 2 SSID's and 2 passwords
StaticJsonBuffer<200> jsonBuffer;

[... 484 lines cut ...]
      connectWiFi();
    }
  }
}
int n = 0;

void loop() {

}
void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}`

cruizg, 2019-02-04:
This is the message:

`:31:12.683 -> Start connection to esp8266
12:31:27.371 -> Connected to wifi
12:31:27.371 -> SSID: esp8266
12:31:27.371 -> IP Address: 192.168.43.18
12:31:27.371 -> signal strength (RSSI):-45 dBm
12:31:27.418 -> -----------------------------------
12:31:27.418 -> ----------Path Exist Test----------
12:31:27.418 -> -----------------------------------
12:31:27.418 -> 
12:31:31.487 -> Path /ESP32_Test is not exist
12:31:31.487 -> 
12:31:31.487 -> -----------------------------------
12:31:31.487 -> ----------Begin Set Test-----------
12:31:31.487 -> -----------------------------------
12:31:31.487 -> 
12:31:33.742 -> ----------Can't set data--------
12:31:33.742 -> REASON: connection refused
12:31:33.742 -> --------------------------------
`

cruizg, 2019-02-05:
Thank you very much, the problem was that since the ESP32 repository was updated (now works from the Arduino IDE and not the manual installation), in the Arduino IDE you have to update the board, do that and now it works fine. (close)

closed 2019-02-05 by cruizg

mobizt, 2019-02-05:
The error caused by low internal memory (RAM) due to both BLE and WiFi (mbedTLS) use large memory. 

Your free heap is only about 32K before start SSL connection which mbedTLS required more memory to work (may be > 100K), which cause SSL connection to failed.

If you write your code that uses more memory, the WiFi connection may failed again. 

I recommend to check the preference for saved WiFi credentials (ssid and password) at the begining and start BLE server only when no preference found or WiFi connection attemp failed. 

Once BLE device was connected and some WiFi config was received then save and restart to begin the device with more memory.