mobizt/Firebase-ESP32 #4: Firebase-ESP32 Have a StreamCallback Function?

cruizg, 2019-02-05:
Do You Have Example With StreamCallBAck Function, because i need do anything when other device write in Firebase Real Time DataBase?
Thanks

mobizt, 2019-02-06:
Ok now I update Firebase-ESP32 library to support set stream callback and remove stream callback functions. Please update and check stream callback example.

cruizg, 2019-02-06:
i try your update but the return is:

4:46:49.174 -> ----------Can't Update data--------
14:46:49.174 -> REASON: connection refused
14:46:49.174 -> --------------------------------
14:46:49.174 -> 
14:46:49.310 -> ----------Can't Update data--------
14:46:49.310 -> REASON: connection refused
14:46:49.310 -> --------------------------------
14:46:49.310 -> 
14:46:49.411 -> ----------Can't get data--------
14:46:49.411 -> REASON: connection refused
14:46:49.444 -> --------------------------------
14:46:49.444 -> 
14:47:10.870 -> ----------Can't update data 1--------
14:47:10.870 -> REASON: connection refused
14:47:10.903 -> --------------------------------

cruizg, 2019-02-06:
when i add :

> Firebase.setStreamCallback(firebaseData, streamCallback1);
>
>  if (!Firebase.beginStream(firebaseData, path))
 > {
>    Serial.println("------Can't begin stream 1 connection------");
 >   Serial.println("REASON: " + firebaseData.errorReason());
>    Serial.println();
>  }

the result is: 
4:47:10.870 -> ----------Can't update data 1--------
14:47:10.870 -> REASON: connection refused
14:47:10.903 -> --------------------------------

but when i comment 

> Firebase.setStreamCallback(firebaseData, streamCallback1);
>
>  if (!Firebase.beginStream(firebaseData, path))
 > {
>    Serial.println("------Can't begin stream 1 connection------");
 >   Serial.println("REASON: " + firebaseData.errorReason());
>    Serial.println();
>  }

that work fine

mobizt, 2019-02-07:
Look like you download the library while I'm editing  it as I seen from your pasted code. 
Actually you should call Firebase.beginStream to begin the stream connection before call Firebase.setStreamCallback (eventhough I write the function comment that can be call before or after).

For better result, use different FirebaseData object for stream and others (get/set/push..) because when you use only one FirebaseData object, WiFi client that use in FirebaseData object will switch between stream and other calls which cause the stream event will not available during get/set/push/update calls.

The switching between stream and other calls due to using the same FirebaseData object can lead to fail connection at the first attempt, you need to call it again.  In this case idle time for letting stream monitoring to work is also required (> 5 seconds).

For connection refuse, SSL WiFi debug print will better show you the cause. Use ESP.getFreeHeap() to check remaining memory may help in memory management to prevent unexpecteded error reboot.

mobizt, 2019-02-07:
This is the better use of stream callback which one FirebaseData object for update or normal Firebase call and other for stream only.

```

#include <WiFi.h>
#include "FirebaseESP32.h"

#define WIFI_SSID "YOUR_WIFI_AP"
#define WIFI_PASSWORD "YOUR_WIFI_PASSWORD"
#define FIREBASE_HOST "YOUR_FIREBASE_PROJECT.firebaseio.com"
#define FIREBASE_AUTH "YOUR_FIREBASE_DATABASE_SECRET"

//Use multiple FirebaseData may lead to low memory and reboot from exception error

//Define FirebaseESP32 data object
FirebaseData firebaseData1;
FirebaseData firebaseData2;

unsigned long sendDataPrevMillis1;
unsigned long sendDataPrevMillis2;

uint16_t count1;
uint16_t count2;

String json;

void streamCallback(streamData data)
{

  Serial.println("-------Stream Data available-------");
  Serial.println("STREAM PATH: " + data.streamPath());
  Serial.println("PATH: " + data.dataPath());
  Serial.println("TYPE: " + data.dataType());
  Serial.print("VALUE: ");
  if (data.dataType() == "int")
    Serial.println(data.intData());
  else if (data.dataType() == "float")
    Serial.println(data.floatData());
  else if (data.dataType() == "string")
    Serial.println(data.stringData());
  else if (data.dataType() == "json")
    Serial.println(data.jsonData());
  Serial.println();
}

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginStream(firebaseData2, "/test/data1"))
  {
    Serial.println("------Can't begin stream 1 connection------");
    Serial.println("REASON: " + firebaseData2.errorReason());
    Serial.println();
  }

  Firebase.setStreamCallback(firebaseData2, streamCallback);

}

void loop()
{

  if (millis() - sendDataPrevMillis1 > 15000)
  {
    sendDataPrevMillis1 = millis();
    count1++;

    json = "{\"data1-1\":" + String(count1) + ",\"data1-2\":" + String(count1 + 1) + ",\"data1-3\":" + String(count1 + 2) + "}";

    if (Firebase.updateNode(firebaseData1, "/test/data1", json))
    {
      Serial.println("----------Update Data 1 result-----------");
      Serial.println("PATH: " + firebaseData1.dataPath());
      Serial.println("TYPE: " + firebaseData1.dataType());
      Serial.print("VALUE: ");
      if (firebaseData1.dataType() == "int")
        Serial.println(firebaseData1.intData());
      else if (firebaseData1.dataType() == "float")
        Serial.println(firebaseData1.floatData());
      else if (firebaseData1.dataType() == "string")
        Serial.println(firebaseData1.stringData());
      else if (firebaseData1.dataType() == "json")
        Serial.println(firebaseData1.jsonData());
      Serial.println("--------------------------------");
      Serial.println();
    }
    else
    {
      Serial.println("----------Can't update data 1--------");
      Serial.println("REASON: " + firebaseData1.errorReason());
      Serial.println("--------------------------------");
      Serial.println();
    }
  }

}

```

mobizt, 2019-02-07:
This is the better use of stream callback which uses one FirebaseData object for update or normal Firebase call and other for stream only.

```

#include <WiFi.h>
#include "FirebaseESP32.h"

#define WIFI_SSID "YOUR_WIFI_AP"
#define WIFI_PASSWORD "YOUR_WIFI_PASSWORD"
#define FIREBASE_HOST "YOUR_FIREBASE_PROJECT.firebaseio.com"
#define FIREBASE_AUTH "YOUR_FIREBASE_DATABASE_SECRET"

//Use multiple FirebaseData may lead to low memory and reboot from exception error

//Define FirebaseESP32 data object
FirebaseData firebaseData1;
FirebaseData firebaseData2;

unsigned long sendDataPrevMillis1;
unsigned long sendDataPrevMillis2;

uint16_t count1;
uint16_t count2;

String json;

void streamCallback(streamData data)
{

  Serial.println("-------Stream Data available-------");
  Serial.println("STREAM PATH: " + data.streamPath());
  Serial.println("PATH: " + data.dataPath());
  Serial.println("TYPE: " + data.dataType());
  Serial.print("VALUE: ");
  if (data.dataType() == "int")
    Serial.println(data.intData());
  else if (data.dataType() == "float")
    Serial.println(data.floatData());
  else if (data.dataType() == "string")
    Serial.println(data.stringData());
  else if (data.dataType() == "json")
    Serial.println(data.jsonData());
  Serial.println();
}

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginStream(firebaseData2, "/test/data1"))
  {
    Serial.println("------Can't begin stream 1 connection------");
    Serial.println("REASON: " + firebaseData2.errorReason());
    Serial.println();
  }

  Firebase.setStreamCallback(firebaseData2, streamCallback);

}

void loop()
{

  if (millis() - sendDataPrevMillis1 > 15000)
  {
    sendDataPrevMillis1 = millis();
    count1++;

    json = "{\"data1-1\":" + String(count1) + ",\"data1-2\":" + String(count1 + 1) + ",\"data1-3\":" + String(count1 + 2) + "}";

    if (Firebase.updateNode(firebaseData1, "/test/data1", json))
    {
      Serial.println("----------Update Data 1 result-----------");
      Serial.println("PATH: " + firebaseData1.dataPath());
      Serial.println("TYPE: " + firebaseData1.dataType());
      Serial.print("VALUE: ");
      if (firebaseData1.dataType() == "int")
        Serial.println(firebaseData1.intData());
      else if (firebaseData1.dataType() == "float")
        Serial.println(firebaseData1.floatData());
      else if (firebaseData1.dataType() == "string")
        Serial.println(firebaseData1.stringData());
      else if (firebaseData1.dataType() == "json")
        Serial.println(firebaseData1.jsonData());
      Serial.println("--------------------------------");
      Serial.println();
    }
    else
    {
      Serial.println("----------Can't update data 1--------");
      Serial.println("REASON: " + firebaseData1.errorReason());
      Serial.println("--------------------------------");
      Serial.println();
    }
  }

}

```

cruizg, 2019-02-07:
Your code work fine!!.. Thank you so much =)

closed 2019-02-08 by mobizt