mobizt/Firebase-ESP32 #5: Question About Library

cruizg, 2019-02-08:
Is there any way to integrate with cloud firestore ?.
Is there any way to send cloud messaging?

thanks a lot.

mobizt, 2019-02-09:
Cloud Firestore data model is more flexible and complex and the implementation is different from real time database.  The authentication is the important and auth tokens (using custom and OAuth2.0 tokens) are not possible in embedded device alone (need external server). It may not support database secret as in real time database (not sure).

For cloud messaging I will add support in the future.

goldenhenry, 2019-02-20:
https://github.com/mobizt/Firebase-ESP32

goldenhenry, 2019-02-20:
https://github.com/mobizt/Firebase-ESP32  
For the library above, when i run the examples "basic". i got error compiling for board ESP32 DEC module, 

"warning: 'typedef' was ignored in this declaration typedef struct FirebaseDataType; "

mobizt, 2019-02-20:
@goldenhenry I just fixed the typedef  issue. Please update.

closed 2019-03-11 by mobizt