mobizt/Firebase-ESP32 #8: What do you call "Path"

LesDeuxM, 2019-02-21:
Hello,
Could you tell me what does the String "path" means ? I tried to let it empty, to put the database name or a variable name, but nothing's working.. I get "Path _xxxx_ is not exist"

Thanks in advance for your help.

mobizt, 2019-02-21:
If you mean function  Firebase.pathExist , path parameter is the full node or key name which is the same as path in other functions. If path e.g. "/any/path/in/db" is not exist, the function returns false. Path in all functions should begin with slash "/".

LesDeuxM, 2019-02-21:
Ok, thanks a lot for your fast answer. I finally solved my issue. In the "FIREBASE_HOST", I had "https://xxxxxxx.firebaseio.com/" instead of "xxxxxxx.firebaseio.com". With the new expression, my path is found and your library works !

closed 2019-02-21 by LesDeuxM