mobizt/Firebase-ESP32 #11: Library not working for Arduino UNO WiFi Rev2

Tommaarleveld, 2019-03-19:
I get an error trying to compile this library for my board. Is there any way I can make this library work for the Arduino UNO WiFi Rev2? It has an ESP32 included on the board:

[image]

This is the error I get:
```
Arduino: 1.8.9 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

Multiple libraries were found for "WiFi.h"
In file included from /Users/tommaarleveld/Documents/Arduino/libraries/Firebase_ESP32_Client/src/FirebaseESP32.h:38:0,
 Used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
 Not used: /Users/tommaarleveld/Documents/Arduino/libraries/WiFiNINA
                 from /Users/tommaarleveld/Documents/Arduino/libraries/Firebase_ESP32_Client/examples/Basic/Basic.ino:6:
/Users/tommaarleveld/Documents/Arduino/libraries/HTTPClientESP32Ex-master/HTTPClientESP32Ex.h:37:24: fatal error: HTTPClient.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.
```

mobizt, 2019-03-19:
The Firebase ESP32 and HTTPClientESP32Ex libraries use WiFi library from ESP32 Core SDK.
UNO WiFi Rev2  can be act as WiFi AP and STA using WiFiNINA library and need to edit the WiFi client in my library to use  WiFiNINA instead of WiFi library from ESP32. 

Unfortunately I don't have UNO WiFi Rev2 board in hand and cannot add support to this board at this time.

I currently order UNO WiFi Rev2 from arduino.cc and may take time to get device and test code, I will update and inform you in this issue.

Tommaarleveld, 2019-03-19:
@mobizt Thank you, I will stand by for updates!

mobizt, 2019-04-11:
Hi Tommaarleveld,

I just finished develop [Firebase Arduino Library for ARM/AVR WIFI Dev Boards based on WiFiNINA](https://github.com/mobizt/Firebase-Arduino-WiFiNINA) which supports Arduino MKR WiFi 1010, Arduino MKR VIDOR 4000 and Arduino UNO WiFi Rev.2 dev boards.

For the limit of program space and memory, I need to remove support for sme special features but it supports standard operation for get, set, push, update, delete and stream database.

closed 2019-04-13 by mobizt