mobizt/Firebase-ESP32 #12: Error on FirebaseESP32.cpp str.erase(std::remove

GuilhermeKROSS, 2019-03-19:
Hi, i'm having the following error message when i compile my project:

...\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp: In member function 'std::__cxx11::string FirebaseESP32::trim(std::__cxx11::string&)':

...\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:2030:13: error: 'remove' is not a member of 'std'

str.erase(std::remove(str.begin(), str.end(), '\r'), str.end());

I tried to comment the lines  2030 and 2031. Doing this i can compile and download. 
Reading values from my Firebase database works ok. But if i try to write a data, i get the fail msg: connection refused. I don't know if this have any relation with the ignored lines on FirebaseESP32.cpp

mobizt Thaks for this amazing work you did and for sharing this great lib with us

mobizt, 2019-03-20:
You're welcome and thank you for let me know.
I will fix this in the next update.

mobizt, 2019-03-20:
I just update to version 2.3.3. Do you still get error when set data?

GuilhermeKROSS, 2019-03-20:
> I just update to version 2.3.3. Do you still get error when set data?

Now everything works fine. Thanks again. Have a good day!

closed 2019-03-22 by mobizt