mobizt/Firebase-ESP32 #13: bool Datatype Missing

breezage, 2019-03-24:
is it possible to add bool data type to this library ?

mobizt, 2019-03-24:
Bool is not support at this time.

Since Firebase RTDB supports only text and number then my library will support only string, json (string), integer (number) and float (number).

If I add support something like boolean, I need to develop special method to store string that prefixed with some extra 5 bytes header e.g. **bool,0** or  **bool,1** to identify it differently than other data types.

If you ever using FirebaseExtend's Firebase-Arduino before, Boolean data type will store as integer 1 for true and 0 for false which no guarantee when you read Boolean data from any path that store as number and evaluate it to Boolean.
Furthermore, if you look at the data in Firebase RTDB panel, you cannot identify the actual data type (bool, integer, float) from number stored at that node.

**I will add support Boolean data type if you don't mind about some extra 5 bytes for each Boolean data.**

There are some exception when you using other client like javascript or other Firebase library to read data from this node, you will get string like **bool,0** or **bool,1**.

breezage, 2019-03-24:
It's okay i kinda got my issue solved so no problem. thank you so much for this library.

closed 2019-03-24 by breezage

mobizt, 2019-03-24:
Now add support Boolean (v 2.3.4).

breezage, 2019-03-26:
omg you added it already that's awesome. great work sir keep it up!