mobizt/Firebase-ESP32 #14: Stream Event Type Patch

GuilhermeKROSS, 2019-04-03:
Hi
I'm having some problem to detect a incoming event using stream functions. I can only detect a stream data when the event type is "PUT". When it's "PATCH" i can't.
I know this because i was using another firebase library where i could choose the event type to get the stream data.
As this library is much more reliable, i'd like to know if there is a way to get PACHT event type on STREAM.

Thanks

mobizt, 2019-04-03:
Actually this library catches put and patch event type already then trig the dataAvailable status and raise the callback function. But the put or patch event may miss in case of unreliable internet connection take place, stream timeout. The library will routinely check event type keep-alive every 30 second to determine stream timeout.

If you want to get actually event type, I will add this feature soon, may be tomorrow.

mobizt, 2019-04-04:
Now I add support the event type already in v 2.3.7 please update.

GuilhermeKROSS, 2019-04-04:
Thanks for you answer.
I have updated my lib, but the problem keeps occuring. I can only read PUT events. I have a ionic 3 framework app which posts the data for my esp32. When the app changes a existing value, or i do it manually on firebase, the Stream can identify the event. But when the app posts a new key, the stream does not catches the data

mobizt, 2019-04-05:
Depends on type of http request.

If your app send patch http request to the stream path, you will get patch from stream.

You can test the library by call Firebase.updateNode which will send patch http request to server.

Please chek from examples [Different_object_stream.ino](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Different_object_stream/Different_object_stream.ino)

mobizt, 2019-04-05:
Patch, put, post are methods in http request from client. Server processes client requests and send back response based on its API. 

Patch is not just you manual update or edit data using Firebase Console. The REST client should send patch http request in server to let server know the connection method or how to process and send  back the proper response payload which generally the payload that client sent.

In your app you need to send http header like this
**GET /uri_that_points_to_endpoint HTTP/1.1**   <- For GET http request
**PUT /uri_that_points_to_endpoint HTTP/1.1**   <- For PUT http request
**POST /uri_that_points_to_endpoint HTTP/1.1**   <- For POST http request
**PATCH /uri_that_points_to_endpoint HTTP/1.1**   <- For PATCH http request

For our Firebase Client Library
**Firebase.setXXX is correspond to http PUT in REST API call.**
**Firebase.pushXXX is correspond to http POST in REST API call.**
**Firebase.updateNode is correspond to http PATCH in REST API call.**
**Firebase.readStream correspond to http GET with "**Accept: text/event-stream**"** header  in REST API call.

mobizt, 2019-04-05:
Patch, put and post are some of many methods used in http request that client send to server. Server processes client request and returns response based on its API. 

**Patch is not just you manual update or edit data using Firebase Console**. The REST client should send patch http request in server to let server know the connection method or how to process and send  back the proper response payload which generally the payload that client sent in case of Firebase RTDB.

In your app you need to send http header like this

**`GET /uri_that_points_to_endpoint HTTP/1.1`**   For GET http request when you want to read data.

**`PUT /uri_that_points_to_endpoint HTTP/1.1`**  For PUT http request when you want to set new data to the existing node or create new node with new data if that nod is not previously existed.

**`POST /uri_that_points_to_endpoint HTTP/1.1`**  For POST http request when you want to append new data to defined node but within the unique (random) parent key.

**`PATCH /uri_that_points_to_endpoint HTTP/1.1`**  For PATCH http request when you want to update any child node under the defined database path. If that path is not exist, it will create new one for you.

For our Firebase Client Library

**`Firebase.setXXX`** <- Perform PUT http request.

**`Firebase.pushXXX`** <- Perform POST http request.

**`Firebase.updateNode`** <- Perform PATCH http request.

**`Firebase.readStream`**  <- Perform GET http request with header that let server know to keep the connection alive.

Please check the example **[Different_object_stream.ino](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Different_object_stream/Different_object_stream.ino)**

Firebase sever will send stream patch event to client when data node or its child at database path get update by data from patch http request. 

Firebase sever will send stream put event to client when data of node or its child at database path replaced by data from other put or post http request or you edit data in console.

closed 2019-04-06 by mobizt