mobizt/Firebase-ESP32 #15: Consecutive Firebase.getInt returns wrong value

nikrath, 2019-04-21:
I have two functions, GetFirstInt and GetSecondInt as follow

```
String GetFirstInt(){
  if (Firebase.getInt(firebaseData, firstPath))
  {
    if(firebaseData.dataType() == "int"){
      return firebaseData.intData();
    }
  }
  return "";
}
String GetSecondInt(){
  if (Firebase.getInt(firebaseData, secondPath))
  {
    if(firebaseData.dataType() == "int"){
      return firebaseData.intData();
    }
  }
  return "";
}
```

These functions are called one after the other in regular intervals of 10 seconds.
Sometimes, the GetSecondInt returns the value of first int that is actually stored in Firebase Database, when it should be returning the value of second int stored in "secondPath".

I even tried to make FirebaseData firebaseData a local variable to these functions instead of a global one. I thought some kind of an overwrite issue might be causing this. But turning it into local variables led the ESP32 to crash for some reason.

Does someone know a fix for this?
Thanks

mobizt, 2019-04-22:
The normal usage of Firebase Data object does not cause problems.
Use Firebase Data object as local variable is not recommend which may lead to unreleased memory used by WiFi core.

Return integer inside String returned function getFirstInt() and getSecondInt() can cause memory leak or give the unexpected result.

You can help me to address the root causes by do the following.
Create example sketch that causes the problems and post it here.
Choose board as Generic ESP32 Module can show the debug message on serial, copy all serial debug message and post it here.

mobizt, 2019-04-22:
Your above code is error and can't compile.
And you're not provide the WiFi connection in setup too.

mobizt, 2019-04-22:
Your above code is error and can't compile due to return integer inside String returned function getFirstInt() and getSecondInt()

And you're not provide the WiFi connection in setup too.

The normal usage of Firebase Data object does not cause problems.

Use Firebase Data object as local variable is not recommend which may lead to unreleased memory used by WiFi core.

You can help me to address the root causes by do the following.

- Create example of working sketch that causes the problems and post it here.
- Choose board as Generic ESP32 Module can show the debug message on serial, copy all serial debug message and post it here.

nikrath, 2019-04-22:
@mobizt Thanks for the response. I have updated the code in the original question, and added some additional information. I will state those here again:

> The flag variable (in CheckFlag function) should return the flag value stored on the Firebase Database in path "/flag".
> BUT Sometimes, the flag (in CheckFlag function) actually returns the value of the date variable stored on firebase database (which should be fetched by prevDate variable in Sync function). But instead prevDate returns 0 (i.e, it fails to fetch a value).

I'll give an example of the above statement. Yesterday, the value stored in my Firebase database at /date was 21, and at /flag was 0.
The code looped and every 10 seconds, it showed the following output:
prevDate = 21 flag = 0
It ran fine for around 8-9 hours. After that, it suddenly shows the following output:
prevDate = 0 flag = 21

I guess prevDate was 0 because it couldn't fetch properly from online (in which case the function FirebaseGetInt returns 0). But what I don't get is why flag variable in code showed the value at /date instead of the value at /flag?

P.S Sorry I couldn't give the credentials of the Firebase Database, it belongs to the company I am currently interning at. But if required, please inform me so and I will create a new dummy database for your testing purpose.

Thanks

mobizt, 2019-04-23:
My library is rather give the exact what type of data store and read from database. 

Please check your free heap and stack size that may lead to heap corruption problem because your application use large of memory for all time.

closed 2019-04-26 by mobizt