mobizt/Firebase-ESP32 #18: two functions tested appear not to be working:append string and getting specific value

usrdes, 2019-05-02:


mobizt, 2019-05-02:
Please explain more detail and example sketch that makes error.

mobizt, 2019-05-02:
How about your setup() and loop() function?

mobizt, 2019-05-02:
How about your serial print out result?

usrdes, 2019-05-02:
Hi Mobizt,
I did not include setup() or loop() because they are OK, in my code. I can see that the WiFi
is setup and talks to firebase OK.  Hope you can copy the above routines in your "basic.ino"
and check it out?

Anyway my serial prinout is included below:

```
Pushing my data to Firebase now 
Done setString /NewData:Kaboom2
/
json
{"data1":"value0","data2":{"_data2":"_value1"}}
{"data1":"value0", "data2":{"_data2":"_value1"}}
Done Pushing my data to Firebase 
Querying Firebase now 
{"NewData":"Kaboom2","data1":"value0","data2":{"_data2":"_value1"}}
Done Querying Firebase 
Querying specific data from Firebase now 
Got NewData as: Kaboom2
{}
Done Querying specific data from  Firebase 
-----------------------------------------------------------------
Pushing my data to Firebase now 
Done pushString /NewData:Kaboom3
/
json
{"data1":"value1","data2":{"_data2":"_value2"}}
{"data1":"value1", "data2":{"_data2":"_value2"}}
Done Pushing my data to Firebase 
Querying Firebase now 
{"NewData":"Kaboom2","data1":"value1","data2":{"_data2":"_value2"}}
Done Querying Firebase 
Querying specific data from Firebase now 
Got NewData as: Kaboom2
{}
Done Querying specific data from  Firebase
-----------------------------------------------------------------
Pushing my data to Firebase now 
Done pushString /NewData:Kaboom4
/
json
{"data1":"value2","data2":{"_data2":"_value3"}}
{"data1":"value2", "data2":{"_data2":"_value3"}}
Done Pushing my data to Firebase 
Querying Firebase now 
{"NewData":"Kaboom2","data1":"value2","data2":{"_data2":"_value3"}}
Done Querying Firebase 
Querying specific data from Firebase now 
Got NewData as: Kaboom2
{}
Done Querying specific data from  Firebase 
-----------------------------------------------------------------
ushing my data to Firebase now 
Done pushString /NewData:Kaboom5
/
json
{"data1":"value3","data2":{"_data2":"_value4"}}
{"data1":"value3", "data2":{"_data2":"_value4"}}
Done Pushing my data to Firebase 
Querying Firebase now 
{"NewData":"Kaboom2","data1":"value3","data2":{"_data2":"_value4"}}
Done Querying Firebase 
Querying specific data from Firebase now 
Got NewData as: Kaboom2
{}
Done Querying specific data from  Firebase
-----------------------------------------------------------------
Pushing my data to Firebase now 
Done pushString /NewData:Kaboom6
/
json
{"data1":"value4","data2":{"_data2":"_value5"}}
{"data1":"value4", "data2":{"_data2":"_value5"}}
Done Pushing my data to Firebase 
Querying Firebase now 
{"NewData":"Kaboom2","data1":"value4","data2":{"_data2":"_value5"}}
Done Querying Firebase 
Querying specific data from Firebase now 
Got NewData as: Kaboom2
{}
Done Querying specific data from  Firebase
-----------------------------------------------------------------
Pushing my data to Firebase now 
Done pushString /NewData:Kaboom7
/
json
{"data1":"value5","data2":{"_data2":"_value6"}}
{"data1":"value5", "data2":{"_data2":"_value6"}}
Done Pushing my data to Firebase 
Querying Firebase now 
{"NewData":"Kaboom2","data1":"value5","data2":{"_data2":"_value6"}}
Done Querying Firebase 
Querying specific data from Firebase now 
Got NewData as: Kaboom2
{}
Done Querying specific data from  Firebase
-----------------------------------------------------------------
Pushing my data to Firebase now 
Done pushString /NewData:Kaboom8
/
json
{"data1":"value6","data2":{"_data2":"_value7"}}
{"data1":"value6", "data2":{"_data2":"_value7"}}
Done Pushing my data to Firebase 
Querying Firebase now 
{"NewData":"Kaboom2","data1":"value6","data2":{"_data2":"_value7"}}
Done Querying Firebase 
Querying specific data from Firebase now 
Got NewData as: Kaboom2
{}
Done Querying specific data from  Firebase 
-----------------------------------------------------------------
Pushing my data to Firebase now 
Done pushString /NewData:Kaboom11
/
json
{"data1":"value9","data2":{"_data2":"_value10"}}
{"data1":"value9", "data2":{"_data2":"_value10"}}
Done Pushing my data to Firebase 
Querying Firebase now 
{"NewData":"Kaboom2","data1":"value9","data2":{"_data2":"_value10"}}
Done Querying Firebase 
Querying specific data from Firebase now 
Got NewData as: Kaboom2
{}
Done Querying specific data from  Firebase 
```

mobizt, 2019-05-02:
What is error and please post your working sketch and your database structure.
From my testing, I don't see the error from your code you provides.

usrdes, 2019-05-02:
I will post full sketch tomorrow, however,
in the listing above the first error is as follows:
1. every time in the loop, the word Kaboom# (where # is 2,3,4,...) is supposed to append to NewData node. For example : 
Iteration 1 in the loop: NewData: Kaboom2, 
Iteration 2 in the loop: NewData: Kaboom2Kaboom3
Iteration 3 in the loop: NewData: Kaboom2Kaboom3Kaboom4 
...
So if you look in the serialprint above, only the first "Kaboom2" is printed as "Got NewData as: Kaboom2".
This means that the NewData did not get appended. 
I also verified this by looking at my database on firebase

2.  when I look for data2:_data2, I expect to see the value as "_value#" as the new value.
However when the serialprint prints out what ever it read back , I see only the following:
"Querying specific data from Firebase now 
Got NewData as: Kaboom2
{}"
So I am not sure if I am using the query filter correctly or if I have an incorrect understanding of
filters or the data structure etc.

mobizt, 2019-05-02:
### First

You declare string obj **s** here

 ```C++

 av2++;  
 String s= "Kaboom"+String(av2);

```

And use it here 1st time

```C++
if( Firebase.setString(firebaseData, "/NewData", s) ) {
   Serial.println("Done setString /NewData:"+s);  // should pust "Kaboom2" to node "NewData"
} else {
   Serial.println(firebaseData.errorReason());
}
```

1st time db

[image]

The 2nd time and later use here

```C++
if (Firebase.pushString(firebaseData, "/NewData", s)) { // should add "Kaboom3" to "Kaboom2"
   Serial.println("Done pushString /NewData:"+s);     // Check firebase data base, to see if 
 } else {                                               // The value of NewData is "Kaboom2Kaboom3"
   Serial.println(firebaseData.errorReason());			// does not seem to be working					
 }
```
The 2nd time result.
[image]

The 3rd and others results

[image]

[image]

Go back to where you declare **s**

 ```C++

 av2++;  
 String s= "Kaboom"+String(av2);

```
And where you push it.
```C++
Firebase.pushString(firebaseData, "/NewData", s)
```

You will always get  **Kaboom#** instead of **Kaboom2Kaboom3Kaboom4Kaboom5....**

___

 ### Second

Again you will see the above image what's happen to **data1** and **data2** nodes.

[image]

Below is your code from functions **query_firebase()** and **GetSpecificData()** that you try to filter the data in the **root node** of your db  or **"/"** 

```C++
if (Firebase.getJSON(firebaseData, "/", query)) // gets the complete data base at "/"
```
or
```C++
if (Firebase.getJSON(firebaseData, "/", query))
```

Which only one **.indexOn** can be apply to specific node ("/" for this case). 

Your query provided two different indexes **data1** i.e. `query.orderBy("data1");` in **query_firebase()** and **data2** i.e. `query.orderBy("data2");` in **GetSpecificData()** which causes the error when make a Firebase API call to server, the error that returns directly from Firebase server will look like the folowing.

**bad request, Index not defined, add \".indexOn\": \"data1\", for path \"/\", to the rules** or
**bad request, Index not defined, add \".indexOn\": \"data2\", for path \"/\", to the rules**

Your database rules should be set one of these
```JS
{
    "rules": {
        ".read": false,
        ".write": false,
        ".indexOn": "data1"
    }
}
```
Or

```JS
{
    "rules": {
        ".read": false,
        ".write": false,
        ".indexOn": "data2"
    }
}
```
When you read the data at the root node path which is the parent of all your data, no need to use query to filter unless you are reading the value of dynamic node  that created under the known parent key.

If you want to filter the value under "/AAA/AAAA/AAAAA"
You may need to set database rules like this.
                      
```JS
{
    "rules": {
        ".read": false,
        ".write": false,
        "AAA": {
            "AAAA": {
                ".indexOn": "AAAAA"
            }
        }
    }
}
```

I recommend to update the library to version 3.0.2.

closed 2019-05-02 by mobizt