mobizt/Firebase-ESP32 #19: Save Data to Internal Memory (SPIFFS) when offline then send it to database after online again

komputerboy, 2019-05-12:
Hi,

I am new using this great ESP32 Firebase library, so i have question. Can i save data to internal memory (like SPIFFS, etc) when there is no WiFi connection detected and  then send it to Firebase after online (Connected to Wifi)?

Thanks

mobizt, 2019-05-12:
From [this example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Retry_and_queue/Retry_and_queue.ino), you can enable **Error Queue** by set max queue in  **Firebase.setMaxErrorQueue**  with number greater than 0.

```C++
//Set maximum queues that allow in Error Queues collection or enable it
Firebase.setMaxErrorQueue(firebaseData, 10);
```
If  Firebase operation failed due to network problem, that process (task) will add to Error Queue collection which you can track its queue id of your last failed Firebase operation.  

All queues in Error Queues collection are waiting for WiFi  for available to run process or task in queues if  `Firebase.beginAutoRunErrorQueue` or `Firebase.processErrorQueue` was call.

If you want to save these queues to file (SD or SPIFFS) 
```C++
Firebase.saveErrorQueue(firebaseData, "/test.txt", QueueStorageType::SPIFFS);
```
Then you can clear all queues if you don't want it to resume (re-process the queue tasks) at this time.

```C++
Firebase.clearErrorQueue(firebaseData);
```

And restore queues from file later with
```C++
 if (Firebase.errorQueueCount(firebaseData, "/test.txt", QueueStorageType::SPIFFS) > 0)
  {
    Firebase.restoreErrorQueue(firebaseData, "/test.txt", QueueStorageType::SPIFFS);
    //Delete queue file if you want
    Firebase.deleteStorageFile("/test.txt", QueueStorageType::SPIFFS);
  }
```

When there are queues already added to Error Queues collection, run the Error Queues with

```C++
//Run Error Queues Automatically with or without callback
Firebase.beginAutoRunErrorQueue(firebaseData, callback);
```
Or run it manually in loop()

```C++
//Run Error Queues Manually
Firebase.processErrorQueue(firebaseData);
```

komputerboy, 2019-05-12:
okay thanks for reply and explanation,

I have tried Your Example about retry and queue and somehow i have this error, Could you please solved it?
```
Arduino: 1.8.9 (Windows 7), Board: "ESP32 Dev Module, Disabled, Default, 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Retry_and_queue:38:15: error: variable or field 'callback' declared void

 void callback (QueueInfo queueinfo){

               ^

Retry_and_queue:38:15: error: 'QueueInfo' was not declared in this scope

Retry_and_queue:38:16: error: variable or field 'callback' declared void

 void callback (QueueInfo queueinfo){

                ^

Retry_and_queue:38:16: error: 'QueueInfo' was not declared in this scope

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\SAMSUNG\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Multiple libraries were found for "FirebaseESP32.h"
 Used: H:\kuliah\arduino\project\libraries\Firebase-ESP32-master
 Not used: H:\kuliah\arduino\project\libraries\arduino_167856
 Not used: H:\kuliah\arduino\project\libraries\Firebase_ESP32_Client
exit status 1
variable or field 'callback' declared void

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
im not changing anything from example code

mobizt, 2019-05-12:
Look like you installed ESP32 Core SDK from many sources (through Git and Boards Manager).

I recommended to ninstall Arduino IDE and delete its folder from program file **C:\Program Files (x86)\Arduino** and also delete folder **C:\Users\SAMSUNG\AppData\Local\Arduino15**

And install ESP32 with [this method](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md)

mobizt, 2019-05-12:
And please update Firebase library to the latest version 3.1.0

mobizt, 2019-05-12:
Ok I look into the error.  In **H:\kuliah\arduino\project\libraries**

Delete these folder **Firebase-ESP32-master**, **arduino_167856**  and **Firebase_ESP32_Client**.

Then re-install Firebase ESP32 Library again (please download from Github to get the latest version).

komputerboy, 2019-05-12:
Okay,

Now i works. Thank You. 
One more question is there any way to increase maximum  queue? maybe with increasing SPIFFS partition size?

mobizt, 2019-05-12:
Maximum is 255. Do you want more than 255 queues?

komputerboy, 2019-05-12:
if it could, Absolutely yes

mobizt, 2019-05-12:
I not recommend because it may bring your device low memory and reset since each queue use internal RAM (not PSRAM) to keep firebase path and data objects for each task (integer, float, double, boolean, string object and blob).

komputerboy, 2019-05-12:
well, its okay if it couldn't. 
Thank You I will close this issue. I think this is the best Firebase ESP32 library

komputerboy, 2019-05-12:
Wait, 
so it is use internal RAM to keep firebase path and data objects, So i cant use this method if i use deep sleep mode in ESP32?

mobizt, 2019-05-12:
Don't understand. This is normal and the only way MCU to work with data.

closed 2019-05-12 by mobizt

komputerboy, 2019-05-12:
okay i will try it, 
is it compatible with Deep Sleep/Light Sleep Mode or not?

mobizt, 2019-05-12:
The library relies on core WiFi library, nothing to say compatible or compatible.

Up on your design and manage to turn off RAM/WiFi for deep sleep or stay connected to AP in light sleep.
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html

usrdes, 2019-05-24:
Hi Mobizt

Is it possible to "setfile" from SPIFF as follows:
`
Firebase.setFile(firebaseData, path, "/myfile.txt", StorageType::SPIFFS)
`
I see your QueueStorageType in your answer above, but it means one has to setup a que.
How about just reading from SPIFFS storage and doing file I/O from firebase?

Thank you

mobizt, 2019-05-24:
I will add this feature in the next library update.