mobizt/Firebase-ESP32 #21: To use as a component of ESP-IDF

ibrahimdanish, 2019-05-15:
I would like to port the library to use it as a component of ESP-IDF. Is it possible to do that?

mobizt, 2019-05-16:
Yes, it's possible.

ibrahimdanish, 2019-05-16:
Good to know that. Could you please guide me a bit how to do that when it comes to the arduino libraries you are using?

mobizt, 2019-05-16:
For [Firebase REST API](https://firebase.google.com/docs/reference/rest/database), you can use tool sush [Postman](https://www.getpostman.com/) to test which is important to understand what's the objective that my code do and how.

For WiFi client, you can create your own WiFi class that use [ESP HTTP Client](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_http_client.html)

Make the interfaces similar to HTTPClientESP32Ex to connect to host, send header and payload and handle connection and status.

For variables store in flash (PROGMEM), you can use const vars which will storre in flash.

My library use C string (char array) and C++ basic string (std::string) which you can use it in esp-idf by include `#include <string>`.  For other Arduino's data types you can use the appopritated data type support in ESP

Other functions almost use C and C++ library which not rely on Arduino.

closed 2019-05-17 by mobizt

erickMbuthia, 2019-05-22:
Thank you Sir for the information.Could you please do the neccesary changes to support the IDF and  post that version on Github?
Am a begginer and i will appreciate if you help me and others.