mobizt/Firebase-ESP32 #23: HELP: firebase.pushJSON with JSONObject created by ArduinoJSON

sarkrui, 2019-05-20:
`const int BUFFER_SIZE = JSON_OBJECT_SIZE(3);`
`StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;`
`JsonObject& logData = jsonBuffer.createObject();`
`logData["dBValue"] = 45.0;`
`logData["time"] = "2019-05-15T15:00:25";`
`logData["buttonState"] = 0;`
`Firebase.pushJSON(firebaseData, data_log + device_id, logData);`

With the above codes, I had such an error. 
> no matching function for call to 'FirebaseESP32::pushJSON(FirebaseData&, StringSumHelper&, ArduinoJson::JsonObject&)'

I have tried using ArduinoJSON v5/v6, but JSONObject isn't supported by firebase.pushJSON. I wonder how other dudes parse variables in the jsonData. From the example, the data is hardcoded though.
`String jsonData = "{\"parent_001\":\"parent 001 text\", \"parent 002\":{\"child_of_002\":123.456}}"; `

mobizt, 2019-05-21:
You need to [print (v5)](https://arduinojson.org/v5/api/jsonobject/printto/) or [serialize (v6)](https://arduinojson.org/v6/example/string/) ArduinoJson object (classs) as Arduino string object or C or C++ string.

sarkrui, 2019-05-21:
Thanks @mobizt ! It works!

### Using ArduinoJson V6
```
  //generated by arduinojson.org/v6/assistant/
  const size_t capacity = JSON_OBJECT_SIZE(3);
  DynamicJsonDocument objectData(capacity);

  objectData["dBValue"] = 45.0;
  objectData["time"] = "2019-05-15T15:00:25";
  objectData["buttonState"] = 0;

  String logData;
  serializeJson(objectData, logData);
  Firebase.pushJSON(firebaseData, data_log + device_id, logData);
```

closed 2019-05-21 by sarkrui

sarkrui, 2019-05-21:
I did manage to pass `int` `float` variables into a JSONObject, however, *only* ints and floats were synchronized, why is that?
```
  //generated by arduinojson.org/v6/assistant/
  const size_t capacity = JSON_OBJECT_SIZE(3);
  DynamicJsonDocument objectData(capacity);

  objectData["dBValue"] = dBValue;  //global variable, float 
  objectData["buttonState"] = buttonState; //global variable, int
  objectData["time"] = formattedDate; //global variable String
  
  String logData;
  serializeJson(objectData, logData);
  Firebase.pushJSON(firebaseData, data_log + device_id, logData);
```
### Firebase Output: 
[image]

mobizt, 2019-05-21:
What do you mean synchronized?

mobizt, 2019-05-21:
Print or serialize your created ArduinoJson object to string and check it on serial print, if something wrong with ArduinoJson please open issue at ArduinoJson Github.

sarkrui, 2019-05-21:
Thanks @mobizt ! I switched to ArduinoJSONV5 and problem solved. 
`not synchronized`, I meant only `int` and `float` were passed, whereas `String` wasn't.
```
void baselinePushV5() {

  const size_t capacity = JSON_OBJECT_SIZE(3);
  DynamicJsonBuffer jsonBuffer(capacity);

  JsonObject& root = jsonBuffer.createObject();
  root["dBValue"] = 56;
  root["buttonState"] = 2;
  root["timeStamp"] = formattedDate;
  root.printTo(serialjson);
  Serial.println(serialjson);
  delay(1000);
}
```
### Serial Outputs
> {"dBValue":56,"buttonState":2,"timeStamp":"2019-05-21T12:10:53"}

Failed to pass String (variable) to JSONObject
https://github.com/bblanchon/ArduinoJson/issues/1002

Proxcentaur, 2020-05-13:
> Thanks @mobizt ! I switched to ArduinoJSONV5 and problem solved.
> `not synchronized`, I meant only `int` and `float` were passed, whereas `String` wasn't.
> 
> ```
> void baselinePushV5() {
> 
>   const size_t capacity = JSON_OBJECT_SIZE(3);
>   DynamicJsonBuffer jsonBuffer(capacity);
> 
>   JsonObject& root = jsonBuffer.createObject();
>   root["dBValue"] = 56;
>   root["buttonState"] = 2;
>   root["timeStamp"] = formattedDate;
>   root.printTo(serialjson);
>   Serial.println(serialjson);
>   delay(1000);
> }
> ```
> 
> ### Serial Outputs
> > {"dBValue":56,"buttonState":2,"timeStamp":"2019-05-21T12:10:53"}
> 
> Failed to pass String (variable) to JSONObject
> [bblanchon/ArduinoJson#1002](https://github.com/bblanchon/ArduinoJson/issues/1002)

How did you pass the root data into Firebase.pushJSON? I did the same thing but still can't pass those data.

mobizt, 2020-05-13:
You can't use ArduinoJson with this library to handle JSON.

The library has built-in JSON parser and builder called FirebaseJson and you should use it instead.

See the example for the usages.

sarkrui, 2020-05-13:
@Proxcentaur Check this out, usage of [FirebaseJSON](https://github.com/mobizt/FirebaseJson).

mobizt, 2020-05-13:
@Proxcentaur Did you read [this](https://github.com/mobizt/Firebase-ESP8266/issues/2#issuecomment-628094708)?

Proxcentaur, 2020-05-14:
Thank you so much, I have read this.

Best regards,
Septia Dini R
Instrumentation Technology
Department of Electrical and Informatics Engineering
Universitas Gadjah Mada