mobizt/Firebase-ESP32 #24: HELP: SPIFFS File I/O to firebase

usrdes, 2019-05-24:
Hi Mobizt

(Sorry if I repeat this question from closed issue "Save Data to Internal Memory ...." , please delete this new issue if this is repeat and you can help in the other thread)

Is it possible to "setfile" from SPIFF as follows:
Firebase.setFile(firebaseData, path, "/myfile.txt", StorageType::SPIFFS)
I see your QueueStorageType in your answer above, but it means one has to setup a que.
How about just reading from SPIFFS storage and doing file I/O from firebase?

Thank you

mobizt, 2019-05-24:
I will add this feature in the next library update. The next update, user can also upload/download file (SD and SPIFFS) to cloud storage data bucket.

usrdes, 2019-05-25:
OK, looking forward to it & also the update to include firestore access :-)

closed 2019-05-25 by usrdes

vanminh1310, 2021-09-01:
Hi. I have saved data to SPIFFS when there is no network connection and when there is network connection I want to read and send data from SPIFFS to Firebase but it is not working can you help me to solve it thank you very much much.
[image]
I did that but it doesn't work

[image]
This is the data I need to send.

mobizt, 2021-09-01:
@vanminh1310 

You data you saved to file contains line breaks which is not valid for JSON.  

You can set temporary data in FirebaseJson object and save (serialized) to file when needed using **toString(file)** and you can read (deserialize) the JSON content from file to FirebaseJson object directly.

This [example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/FirebaseJson/examples/Stream/File/File.ino) shows how to read JSON content from file and save JSON content in FirebaseJson object to file.

The example above required the latest version library update.

vanminh1310, 2021-09-02:
I have tried the following code but it doesn't work. Sorry for the inconvenience, can you give me a solution. I thank you so much.
[image]

mobizt, 2021-09-02:
@vanminh1310 

Did you understand "line breaks" I mentioned?

It means "new line" character or "enter" which should not contains in the string of path or key and value of data

vanminh1310, 2021-09-02:
Thank you, I will watch it again

mobizt, 2021-09-02:
@vanminh1310 

The first method seems to be ok but the second method is not ok as it uses println which append new line to your data.

The first method use printTo of that JSON library which I never try that, you should check if it appended new line, it can't use.

The example I mentioned  above is easier to serialize or deserialize JSON to/from File https://github.com/mobizt/Firebase-ESP32/blob/master/examples/FirebaseJson/examples/Stream/File/File.ino

You should learn how to use FirebaseJson library which is the built-in class that manage the JSON data for the Firebase client library.

Other JSON libraries other than FirebaseJson are not supported by this Firebase library.

FirebaseJson is now powered by [cJSON](https://github.com/DaveGamble/cJSON) the best and light weight and efficient JSON library in C.

vanminh1310, 2021-09-02:
Thank you very much