mobizt/Firebase-ESP32 #26: HELP: Cloud_Messaging.ino can the tokens be MAC address

usrdes, 2019-06-03:
Hi,
In Cloud_Messaging.ino example :
```

#define FIREBASE_FCM_DEVICE_TOKEN_1 "RECIPIENT_DEVICE_TOKEN"
#define FIREBASE_FCM_DEVICE_TOKEN_2 "ANOTHER_RECIPIENT_DEVICE_TOKEN"

```
What are Device Tokens? 
How does one get them from Firebase? 
Are they assigned by the user or firebase? 
If they are assigned by user, are they arbitrary strings? Mac addresses?
If they are Mac addresses, then two ESP32's located on different WiFi networks should be able
to send messages to each other?

I was unable to figure it out by reading the firebase docs, or searching online.
Please help

mobizt, 2019-06-03:
The device token or App instant ID token is the unique Identifier that issues from Firebase SDK in Firebase client apps included Android, iOS and web apps.

You need to have these app or create your own by using [this Android quick start app](https://github.com/firebase/quickstart-android/tree/master/messaging), [setup Android/Web in console](https://firebase.google.com/docs/android/setup), [setup Android app](https://firebase.google.com/docs/cloud-messaging/android/client) or or [web quick start app](https://github.com/firebase/quickstart-js/tree/master/messaging), [web setup](https://firebase.google.com/docs/cloud-messaging/js/client). 

The id token for mobile app will be created for every app installed and can be renew when app was uninstalled and reinstall. For web app, it depends on how you create the token.

usrdes, 2019-06-04:
Hi Mobizt, thank you for taking the time to respond. Hope you can answer some more:

So when I set up an android app, under Settings->General, the App Id is listed as:
1:15nnnnnnnnn3:android:12345678987654321 (modified)
The "token" as shown above starts with a "1:". So is this part of the token, or should I remove it?
How about the ":android:" part? Should I keep that part as well?
Or is the token only the numbers between "1:" and ":android:"?

How about if I use an ESP32 to ESP32 messaging via cloud messaging? Can I still use the above tokens, or is cloud messaging strictly between android/ios/web to/from device? 
Does your library support such device to device messaging?

mobizt, 2019-06-04:
That App ID is not token, the token which sometimes call App ID token or App Instant ID token, it is encoded string like this 

eQyBqxjMJXc:APA91bFoWvte0_BEXsASHw7PEna5Y8tKmrt8uL-3N2dPyFvL7YbdPyjTJBmBZJJiznNA2fjdULooPuhtgqTE8WsMSt-DTgYo9SXvW02jr2G4pXXyddEWwEWGXXQ21oSpbiL3Q0a21bCx

This token generated by app that use the library from Firebase Messaging SDK when develop the app and this token belong to any app instance which can be changed as user re-install app or use the app for the first time.

The REST client, as this library can not receive the message as API is not support it. Then messaging between ESP32 and ESP32 is not possible.

Any app that created with Firebase SDK can send the message to each other or to group or to topic that any app was subscribed.

If you want to send and recieve data/message between ESP32, you can use stream in Firebase RTDB too  but MQTT can do it better.

usrdes, 2019-06-04:
OK, Thank you very much

closed 2019-06-04 by usrdes