mobizt/Firebase-ESP32 #29: HELP How to send JSON string to firebase

mixaobo, 2019-07-11:
here is my code, i can set each value but need totaly 4 times, i know there is a setJSON but i cant get it work. 
here is my code 
[My code and error](https://imgur.com/DuhcRzU)

mobizt, 2019-07-11:
Function setJSON require json serialized string.

Please remember that JSON is only key-value paired notation which can be handle or manage differently depends on system and parser.

The ArduinoJson handle and manage it by its own class (object), form create, parse, to serialize, and it has it own function that return to string directly as stansard c++ string or C chars array for you to use.

ArduinoJson v5 provide **printTo** function to return string from its JSON object.

```C++
String myString;
object.printTo(myString);

//Now use myString in setJSON function

Firebase.setJSON(firebaseData, path1, myString);
```

https://arduinojson.org/v5/api/jsonobject/printto/

mixaobo, 2019-07-11:
> Function setJSON require json serialized string.
> 
> Please remember that JSON is only key-value paired notation which can be handle or manage differently depends on system and parser.
> 
> The ArduinoJson handle and manage it by its own class (object), form create, parse, to serialize, and it has its own functions that return to string directly as stansard c++ string or C chars array for you to use.
> 
> ArduinoJson v5 provide **printTo** function to return string from its JSON object.
> 
> ```c++
> String myString;
> object.printTo(myString);
> 
> //Now use myString in setJSON function
> 
> Firebase.setJSON(firebaseData, path1, myString);
> ```
> 
> https://arduinojson.org/v5/api/jsonobject/printto/

> Function setJSON requires json serialized string.
> 
> Please remember that JSON is only key-value paired notation which can be handle or manage differently depends on system and parser.
> 
> The ArduinoJson handle and manage it by its own class (object), form create, parse, to serialize, and it has its own functions that return to string directly as stansard c++ string or C chars array for you to use.
> 
> ArduinoJson v5 provides **printTo** function to return string from its JSON object.
> 
> ```c++
> String myString;
> object.printTo(myString);
> 
> //Now use myString in setJSON function
> 
> Firebase.setJSON(firebaseData, path1, myString);
> ```
> 
> https://arduinojson.org/v5/api/jsonobject/printto/

sorry for noob question, since i use arduinojson v6. are there anyway to use printTo. they just use serializeJson

mobizt, 2019-07-11:
Read this for how to print as string
https://arduinojson.org/v6/example/string/

closed 2019-07-11 by mixaobo