mobizt/Firebase-ESP32 #30: BUG -- Firebase set requestes does not works anywhere.

sapuz1508, 2019-07-22:
I made some changes to my code and discoverd something strage.
```

    Serial.begin(115200); 

    pinMode(READY_LED, OUTPUT);

    pinMode(ACTIVITY_LED, OUTPUT);

    pinMode(ERROR_LED, OUTPUT);

    pinMode(DEFAULT_BUTT, INPUT);

    pinMode(SERVICE_BUTT, INPUT);

    preferences.begin("my-app", false);

    SPIFFS.begin(true);

    ssid=preferences.getString("ssid"); //take my wifi data

    password=preferences.getString("password");

    utente=preferences.getString("utente");

    reconnectionTime = preferences.getInt("reconnectionTime", 60000);

    delayReconnectionTime = preferences.getInt("reconnectionDelay", 60000);

    Serial.print("SSID: "); Serial.println(ssid);

    Serial.print("UTENTE: "); Serial.println(utente);

    Serial.print("PASSWORD: "); Serial.println(password);

    Serial.print("MAC: "); Serial.println(preferences.getString("macAddress"));

    if(ssid != "" && !preferences.getBool("reset"))

    {

        preferences.putBool("reset", false); //not influent

        

		createThread(&reset, "reset", &resetTask, 1); //not influent

	

        if(utente == "") //decide if we're about to connect to a standard wifi or a peap one and connect

	    connectWiFi();

	else

	    connectPEAP();

        connectMQTT();

    }

    else

    {

        preferences.putBool("reset", true);

        createAP();

    }
    
    if(WiFi.status() == WL_CONNECTED) //if i'm connected to wifi ill connect to firebase

    {
		Serial.println("Connecting to firebase");

		Firebase.begin("SomeAddress", "SomeSecret");

		Firebase.setMaxRetry(firebase, 3);

		Firebase.setMaxErrorQueue(firebase, 100);
		Firebase.reconnectWiFi(true);

    }

    afterAlive = 0;

    PubAliveWiFi(); //in this function i call Firebase.setInt() onto a node that already exists but it doesnt make anything. If i put the Firebase.setInt() into an if statement if goes in the true path.
```
This is quite strange, i expect that note to change.

Arduino 1.8.9
ESP-32 WROOM 1.0.2 SDK Version.

mobizt, 2019-07-22:
You should print the debug from firebaseData.errorReason() and post your problem line of code. Information you give is not enough.

sapuz1508, 2019-07-22:
@mobizt
As I told you if I put the Firebase.setInt() into an if it returns true so according to your examples it doesn’t return errors

mobizt, 2019-07-22:
There's nothing to help because you don't show the code that you think it's not work.

It's depends on your program design, flow of your program and third party library you used.

Test each task for each lib separately then add it one by one to your program, test, debug and add another.

mobizt, 2019-07-22:
If you read [this](https://github.com/mobizt/Firebase-ESP32#store-data), you will see the true status for each Firebase function call was set by **server returned HTTP header status 200 AND matches return payload data type**. 

If it returns true, it make sure that your data was successfully stored at server based on server responses.

closed 2019-07-22 by mobizt

sapuz1508, 2019-07-23:
Wait, let me explain

Heres a part of the code i sent yesterday.

```
 if(ssid != "" && !preferences.getBool("reset"))

    {

        preferences.putBool("reset", false); //not influent

        

		createThread(&reset, "reset", &resetTask, 1); //not influent

	

        if(utente == "") //decide if we're about to connect to a standard wifi or a peap one and connect

	    connectWiFi();

	else

	    connectPEAP();

        connectMQTT();

    }

    else

    {

        preferences.putBool("reset", true);

        createAP();

    }
    
    if(WiFi.status() == WL_CONNECTED) //if i'm connected to wifi ill connect to firebase

    {
		Serial.println("Connecting to firebase");

		Firebase.begin("SomeAddress", "SomeSecret");

		Firebase.setMaxRetry(firebase, 3);

		Firebase.setMaxErrorQueue(firebase, 100);
		Firebase.reconnectWiFi(true);

    }

    afterAlive = 0;

    PubAliveWiFi(); //this function is: 

    if(preferences.getBool("otaCompleted", false))
    {
        Serial.println("OTA Completed successfully, reboot done");
        Firebase.setBool(firebase, "/devices/" + mac + "/update/ota", false); //this works
        preferences.putBool("otaCompleted", false);
    }
   
    }

    PubAliveWiFi()
    {
        if(lastPub + afterAlive < millis() && !offlineMode)
        {
            int WiFiSignal = WiFi.RSSI()+100;

            WiFiSignal *= 2;

            if (WiFiSignal > 100)
                WiFiSignal = 100;

            bool returned = Firebase.setInt(firebase, "/device/" + mac + "/wifi", WiFiSignal); //this does not work even though it returns true.
            Serial.print("Firebase error: "); Serial.println(firebase.errorReason()); //Prints: "Firebase error: "
            Serial.print("Firebase return: "); Serial.println(returned); //Prints: "Firebase return: 1"
            
            lastPub = millis();
            afterAlive = 120000;
    }

```

so, as i said, sometimes it works and sometimes it doesnt even though the code says everything is ok..
with "doesnt works" i mean the the values in firebase are not changed.
@mobizt

mobizt, 2019-07-23:
Try to use `Serial.println (firebase.payload());` after the code that you set int to see what payload response returned from server.

Test and post the serial printed result when you get the issue.

```C++
bool returned = Firebase.setInt(firebase, "/device/" + mac + "/wifi", WiFiSignal); //this does not work even though it returns true.

Serial.println (firebase.payload());
```

sapuz1508, 2019-07-23:
It actually works now.. i dont really understand why.. but it sure does.
And the payload now returns the value I wrote into the DB.
Thank you anyway.

mobizt, 2019-07-23:
I also see that you use variable **"mac"** that I don't see how it's assignment.
You should debug print the path that pass to Firebase function to make sure that it contains the value as you expeced  not modified or change.