mobizt/Firebase-ESP32 #31: HELP Which protocol does esp32 use for pushing data to Firebase?

mixaobo, 2019-07-26:
Sorry this question isnt about your bug. This is question for which protocol do you use to push data.
[In the case of Firebase Realtime Database, clients will be connected to the database and will maintain an open bidirectional connection via websockets](https://www.freecodecamp.org/news/iot-prototyping-with-firebase-doing-more-with-less-2f5c746dac8b/) i found this one said Firebase use websocket to receive or push data to client. in you code, i see you use HTTP protocol. i have confuse a bit. 

Could you confirm which one is correct?

mobizt, 2019-07-26:
Google does not provide the websocket end point to connect to Firebase server directly. They provide the APIs (in SDK) for all their Firebase products included RTDB, for various platforms e.g JavaScript, Android, iOS, REST etc.  

Only [REST APIs](https://firebase.google.com/docs/reference/rest/database) are suitable for internet capable with limited resource which is not depending on platform or OS. 

That's is the only approach where we use to communicate to Firebase using HTTP request to the REST API end point.

closed 2019-07-27 by mixaobo

mrdc, 2019-10-21:
@mixaobo 
>  i see you use HTTP protocol.

> we use to communicate to Firebase using HTTP request

BTW data is send via HTTPS to Firebase and back.