mobizt/Firebase-ESP32 #33: Connection refused, even though I see no reason why it shouldn't be.

Mutad0r, 2019-09-02:
Using NodeMCU. Trying the "basic" example.

So I tried googling a lot and learned from some other posts made by you to check the serial error messages and  Iindeed do get:

> [E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for eden-gardener.firebaseio.com/
> [E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -1

This should mean that there's an internet connection error, but I made sure that that's not the problem. I have made sure that the AP has internet and I have tried using a different AP.

I also tried using the instructions in the readme to implement firebase in the WIP project where I intend to use this library. I get the same error there, but more importantly, that project already has an implementation of getting time from the NTP, which works fine, so I do have connection to the internet.

Furthermore, I initially had tried to use firebase through this instruction: [https://iotdesignpro.com/projects/iot-controlled-led-using-firebase-database-and-esp32](url)
Which worked and I could get and set data on the realtime database.

I wanted to migrate to your library because the "IOXhop_FirebaseESP32" library was not quite cutting it and it was last updated 2 years ago.

The only thing I could think of is that the paths used in your example do not exist in my database, since I couldn't really understand how the paths is supposed to work (where is the root) and I hoped that when setting a value, if the variable didn't exist, it would create it.

I have no clue what to do at this point other than revert back to the IOXhop library, and I really don't want to do that.

mobizt, 2019-09-02:
Look like you provide invalid Firebase host. For example [this](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Basic/Basic.ino#L19) which means you provide the wrong FIREBASE_HOST like this **eden-gardener.firebaseio.com/** instead of  **eden-gardener.firebaseio.com**. 

Someone also open the same issue, check [this](https://github.com/mobizt/Firebase-ESP8266/issues/5#issuecomment-479474668).

Connecttion refused is returned directly from Firebase server and showed that somethings are wrong to make connection (invalid url, endpoint and authentication). 

No matter what the path use in examples is, that path will be created in your project when you run examples.

closed 2019-09-02 by mobizt

mobizt, 2019-09-02:
### Depending on your requirement, other Firebase library you mention is total different in API and features and also not compatible with this library.

Mutad0r, 2019-09-02:
Well that was that indeed. It totally works now. Thanks for a fast reply and good job spotting the error in my code from just the error message without having to see my code at all. I had figured that the address was the same, so just copying would work fine, since it had worked previously.