mobizt/Firebase-ESP32 #34: RAM usage improvement?

mrdc, 2019-10-24:
Hello,

I'm using the Firebase library in my project on ESP32 Wroom, it doesn't have additional PSRAM, so there is not too much free RAM to use. 
What I've noticed is that my ESP32 Free Heap is 51292 after boot. I've compared Free Heap to the library IOXhop_FirebaseESP32 which gives me ~94700 after boot with the same code as used with your library.
Do you think there is space for RAM usage improvement?

mobizt, 2019-10-24:
It's strange and unbelievable.
For example if you use two FirebaseData object, the free heap is about 160-170k.
Something wrong with your code? 
You should share your code.

mrdc, 2019-10-24:
I'm using BLE - it uses a lot of RAM.

mobizt, 2019-10-24:
You should not use BLE together with WiFi because it use the same radio.
You need to disable BLE and reboot to use WiFi only get more free heap.

mrdc, 2019-10-24:
BTW it's possible to use at the same time even if it uses the same radio.

mobizt, 2019-10-24:
No

closed 2019-10-24 by mobizt

mrdc, 2019-10-24:
I know about mbed memory usage. In may case it works OK.

mobizt, 2019-10-24:
Are you sure for SSL/TLS?

mrdc, 2019-10-24:
How can I check it? What I see is that ESP32 connects to 443 port.

mobizt, 2019-10-24:
You can test to use WiFiClientSecure to connect to some server  via ssl (use https in server url or host).  It uses a lot of ram in SSL handshake and decrypt/encrypt exchanged data.  

I'm not recommended to use both BLE and WiFi at the same time for high WiFi priority tasks as stream which it needs connection to stay alive.