mobizt/Firebase-ESP32 #35: HELP. connection refused

cocohgo, 2019-10-25:
`#include <FirebaseESP32.h>
#include <FirebaseESP32HTTPClient.h>
#include <FirebaseJson.h>
#include <jsmn.h>

const char* ssid = "Lab_118";
const char* password = "";
#include "img_converters.h"

String FIREBASE_HOST = "fotos-81fec.firebaseio.com";
String FIREBASE_AUTH = "RU0EZbsEhAZHdk8KhZXjsCLGGTuRe4Wp5vBZbMXI";

FirebaseData firebaseData;

#include <WiFi.h>
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"
#include "Base64.h"

#include "esp_camera.h"

#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

#define XCLK_FREQUENZ  20000000
#define RESOLUTION  FRAMESIZE_XGA
#define BILDNAME "LED_CAM_"

static camera_config_t camera_config = {
    .pin_pwdn  = PWDN_GPIO_NUM,
    .pin_reset = RESET_GPIO_NUM,
    .pin_xclk  = XCLK_GPIO_NUM,
    .pin_sscb_sda = SIOD_GPIO_NUM,
    .pin_sscb_scl = SIOC_GPIO_NUM,

    .pin_d7 = Y9_GPIO_NUM,
    .pin_d6 = Y8_GPIO_NUM,
    .pin_d5 = Y7_GPIO_NUM,
    .pin_d4 = Y6_GPIO_NUM,
    .pin_d3 = Y5_GPIO_NUM,
    .pin_d2 = Y4_GPIO_NUM,
    .pin_d1 = Y3_GPIO_NUM,
    .pin_d0 = Y2_GPIO_NUM,
    .pin_vsync = VSYNC_GPIO_NUM,
    .pin_href = HREF_GPIO_NUM,
    .pin_pclk = PCLK_GPIO_NUM,

    //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = XCLK_FREQUENZ,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_JPEG,//YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = RESOLUTION,   //QQVGA-QXGA Do not use sizes above QVGA when not JPEG

    .jpeg_quality = 12, //0-63 lower number means higher quality
    .fb_count = 1 //if more than one, i2s runs in continuous mode. Use only with JPEG
};

esp_err_t camera_init(void);
esp_err_t camera_capture(void);
void setSensordaten(void);

void setup() { 
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();
  Serial.println("ssid: " + (String)ssid);
  Serial.println("password: " + (String)password);
  
  WiFi.begin(ssid, password);

  long int StartTime=millis();
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      if ((StartTime+10000) < millis()) break;
  } 

  if (WiFi.status() == WL_CONNECTED) {
    char* apssid = "ESP32-CAM";
    char* appassword = "12345678";         //AP password require at least 8 characters.
    Serial.println(""); 
    Serial.print("Camera Ready! Use 'http://");
    Serial.print(WiFi.localIP());
    Serial.println("' to connect");
    WiFi.softAP((WiFi.localIP().toString()+"_"+(String)apssid).c_str(), appassword);            
  }
  else {
    Serial.println("Connection failed");
    return;
  } 
  
  esp_err_t err = esp_camera_init(&camera_config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    delay(1000);
    ESP.restart();
  }
  
  setSensordaten();
  sensor_t *s = esp_camera_sensor_get();
  
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  Firebase.setMaxRetry(firebaseData, 3);
  Firebase.setMaxErrorQueue(firebaseData, 30); 
  Firebase.enableClassicRequest(firebaseData, true);

 // String jsonData = "{\"photo\":\"" + Photo2Base64() + "\"}";
  FirebaseJson json;
  json.set("{\"photo\":\"" + Photo2Base64() + "\"}");
  String photoPath = "/esp32cam";
  if (Firebase.pushJSON(firebaseData, photoPath, json)) {
    Serial.println(firebaseData.dataPath());
    Serial.println(firebaseData.pushName());
    Serial.println(firebaseData.dataPath() + "/"+ firebaseData.pushName());
  } else {
    Serial.println(firebaseData.errorReason());
  }
} 
 
void loop() { 
  delay(10000);
}

String Photo2Base64() {
    setSensordaten();
    sensor_t *s = esp_camera_sensor_get();
    camera_fb_t * fb = NULL;
    fb = esp_camera_fb_get();  
    if(!fb) {
      Serial.println("Camera capture failed");
      return "";
    }
  
    String imageFile = "data:image/jpeg;base64,";
    char *input = (char *)fb->buf;
    char output[base64_enc_len(3)];
    for (int i=0;i<fb->len;i++) {
      base64_encode(output, (input++), 3);
      if (i%3==0) imageFile += urlencode(String(output));
    }

    esp_camera_fb_return(fb);
    
    return imageFile;
}

//https://github.com/zenmanenergy/ESP8266-Arduino-Examples/
String urlencode(String str)
{
    String encodedString="";
    char c;
    char code0;
    char code1;
    char code2;
    for (int i =0; i < str.length(); i++){
      c=str.charAt(i);
      if (c == ' '){
        encodedString+= '+';
      } else if (isalnum(c)){
        encodedString+=c;
      } else{
        code1=(c & 0xf)+'0';
        if ((c & 0xf) >9){
            code1=(c & 0xf) - 10 + 'A';
        }
        c=(c>>4)&0xf;
        code0=c+'0';
        if (c > 9){
            code0=c - 10 + 'A';
        }
        code2='\0';
        encodedString+='%';
        encodedString+=code0;
        encodedString+=code1;
        //encodedString+=code2;
      }
      yield();
    }
    return encodedString;
}

void setSensordaten()
{
  size_t res = 0;
  sensor_t * s = esp_camera_sensor_get();

//  res = s->set_framesize(s, (framesize_t)val);          // Framesize      FRAMESIZE_(QQVGA, HQVGA, QVGQ, CIF, VGA, SVGA, XGA, SXGA, UXGA)
  res = s->set_quality(s, 10);                          // JPEG Quality   (10...63)  10
  res = s->set_contrast(s, 0);                          // Kontrast       (-2...+2)   0
  res = s->set_brightness(s, 0);                        // Helligkeit     (-2...+2)   0
  res = s->set_saturation(s, 0);                        // Farbsättigung  (-2...+2)   0 
  res = s->set_gainceiling(s, (gainceiling_t) GAINCEILING_2X);      // Verstärkung    GAINCEILING_(2x, 4x, 8x, 16x, 32x, 64x, 128x)   2x
  res = s->set_colorbar(s, 0);                          // Farbbalken     (off/on)   off 
  res = s->set_whitebal(s, 1);                          // Weißbalance    (off/on)    on
  res = s->set_gain_ctrl(s, 1);                         // AGC            (off/on)    on
  res = s->set_exposure_ctrl(s, 1);                     // AEC            (off/on)    on               
//  res = s->set_hmirror(s, val);                         // H-Mirror       (off/on) 
//  res = s->set_vflip(s, val);                           // V-Flip         (off/on) 
//  res = s->set_awb_gain(s, val);                        // Verstärkung AWB
//  res = s->set_agc_gain(s, val);                        // Verstärkung AGC
//  res = s->set_aec_value(s, val);                       // Wert AEC       (0...1200)  
//  res = s->set_aec2(s, val);                            // Wert AEC2
  res = s->set_dcw(s, 1);                             // Downsize       (off/on)    on  
  res = s->set_bpc(s, 0);                             //                (off/on)   off
  res = s->set_wpc(s, 1);                             //                (off/on)    on
  res = s->set_raw_gma(s, 1);                         // RAW-Gamma      (off/on)    on
  res = s->set_lenc(s, 1);                            // Linsenkorr.    (off/on)    on 
//  res = s->set_special_effect(s, val);                  // Special Effekt 
//  res = s->set_wb_mode(s, val);                         // WB Mode         (Auto/Sunny/..) 
  res = s->set_ae_level(s, 0);                        // AE Level        (-2...+2)    0
}
`

cocohgo, 2019-10-25:
With the update of the library today, this sketch can´t connect with firebase database

mobizt, 2019-10-25:
To set JSON string, use 
```C++
json.setJsonData("{\"photo\":\"" + Photo2Base64() + "\"}");
```
instead of
```C++
json.set("{\"photo\":\"" + Photo2Base64() + "\"}");
```

cocohgo, 2019-10-26:
Thanks, that was the solution

closed 2019-10-26 by cocohgo