mobizt/Firebase-ESP32 #36: help:with accessing particular json path data.

vamsikrishnaA9, 2019-10-26:
hi there,
thanks for the library.
i need to access the particular json key data.
is there any way to access the particular key data after getting the whole json data.
thank you

mobizt, 2019-10-26:
Make sure you update library to the latest version. I recommended to download from git version.

You can use **FirebaseJsonData** to get the parse result like this.

Please refer to the doc [here](https://github.com/mobizt/Firebase-ESP32#parse-create-and-edit-json-objects) 
```C++

//Declare FirebaseJsonData object to get the parse result, similar to FirebaseData object
FirebaseJsonData jsonData;

json.get(jsonData, "unit/temp2");

if (jsonData.success)
{
  //Print type of parsed data e.g string, int, double, bool, object, array, null and undefined
  Serial.println(jsonData.type);
  //Print its content e.g.string, int, double, bool whereas object, array and null also can access as string
  Serial.println(jsonData.stringValue);
  //Serial.println(jsonData.intValue);
  //Serial.println(jsonData.boolValue);
  //Serial.println(jsonData.doubleValue);
}

```

vamsikrishnaA9, 2019-10-26:
It works . Thank you

closed 2019-10-26 by vamsikrishnaA9