mobizt/Firebase-ESP32 #38: HELP. Upload photo bigger than QVGA

cocohgo, 2019-10-30:
Hi. I have this code to upload photos taken with esp32 cam and OV2640, but it only uploads a QVGA size.
Do you know why and how can I send a larger size?

`const char* ssid = "Lab_118";
const char* password = "";
#include "img_converters.h"

String FIREBASE_HOST = "fotos-81fec.firebaseio.com";
String FIREBASE_AUTH = "RU0EZbsEhAZHdk8KhZXjsCLGGTuRe4Wp5vBZbMXI";

//Library: https://github.com/mobizt/Firebase-ESP32
#include "FirebaseESP32.h"
FirebaseData firebaseData;

#include <WiFi.h>
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"
#include "Base64.h"

#include "esp_camera.h"

//CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

void setup() { 
    
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();
  Serial.println("ssid: " + (String)ssid);
  Serial.println("password: " + (String)password);
  
  WiFi.begin(ssid, password);

  long int StartTime=millis();
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      if ((StartTime+10000) < millis()) break;
  } 

  if (WiFi.status() == WL_CONNECTED) {
    char* apssid = "ESP32-CAM";
    char* appassword = "12345678";         //AP password require at least 8 characters.
    Serial.println(""); 
    Serial.print("Camera Ready! Use 'http://");
    Serial.print(WiFi.localIP());
    Serial.println("' to connect");
    WiFi.softAP((WiFi.localIP().toString()+"_"+(String)apssid).c_str(), appassword);            
  }
  else {
    Serial.println("Connection failed");
    return;
  } 

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  config.frame_size = FRAMESIZE_QVGA;  // UXGA|SXGA|XGA|SVGA|VGA|CIF|QVGA|HQVGA|QQVGA
  config.jpeg_quality = 10;
  config.fb_count = 1;
  
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    delay(1000);
    ESP.restart();
  }
  
  setSensordaten();
  sensor_t *s = esp_camera_sensor_get();
  
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  Firebase.setMaxRetry(firebaseData, 3);
  Firebase.setMaxErrorQueue(firebaseData, 30); 
  Firebase.enableClassicRequest(firebaseData, true);

  FirebaseJson json;
  json.setJsonData("{\"photo\":\"" + Photo2Base64() + "\"}");
  String photoPath = "/esp32-cam";
  if (Firebase.pushJSON(firebaseData, photoPath, json)) {
    Serial.println(firebaseData.dataPath());
    Serial.println(firebaseData.pushName());
    Serial.println(firebaseData.dataPath() + "/"+ firebaseData.pushName());
  } else {
    Serial.println(firebaseData.errorReason());
  }
} 
 
void loop() { 
  delay(10000);
}

String Photo2Base64() {
    setSensordaten();
    sensor_t *s = esp_camera_sensor_get();
    camera_fb_t * fb = NULL;
    fb = esp_camera_fb_get();  
    if(!fb) {
      Serial.println("Camera capture failed");
      return "";
    }
  
    String imageFile = "data:image/jpeg;base64,";
    char *input = (char *)fb->buf;
    char output[base64_enc_len(3)];
    for (int i=0;i<fb->len;i++) {
      base64_encode(output, (input++), 3);
      if (i%3==0) imageFile += urlencode(String(output));
    }

    esp_camera_fb_return(fb);
    
    return imageFile;
}

//https://github.com/zenmanenergy/ESP8266-Arduino-Examples/
String urlencode(String str)
{
    String encodedString="";
    char c;
    char code0;
    char code1;
    char code2;
    for (int i =0; i < str.length(); i++){
      c=str.charAt(i);
      if (c == ' '){
        encodedString+= '+';
      } else if (isalnum(c)){
        encodedString+=c;
      } else{
        code1=(c & 0xf)+'0';
        if ((c & 0xf) >9){
            code1=(c & 0xf) - 10 + 'A';
        }
        c=(c>>4)&0xf;
        code0=c+'0';
        if (c > 9){
            code0=c - 10 + 'A';
        }
        code2='\0';
        encodedString+='%';
        encodedString+=code0;
        encodedString+=code1;
        //encodedString+=code2;
      }
      yield();
    }
    return encodedString;
}

void setSensordaten()
{
  size_t res = 0;
  sensor_t * s = esp_camera_sensor_get();

  res = s->set_quality(s, 10);                          
  res = s->set_contrast(s, 0);                          
  res = s->set_brightness(s, 0);                      
  res = s->set_saturation(s, 0);                      
  res = s->set_colorbar(s, 0);                         
  res = s->set_whitebal(s, 1);                         
  res = s->set_gain_ctrl(s, 1);                         
  res = s->set_gainceiling(s, (gainceiling_t) GAINCEILING_2X);   
  res = s->set_exposure_ctrl(s, 1);                    
  res = s->set_awb_gain(s, 1);                        
  res = s->set_aec2(s, 0);                           
  res = s->set_dcw(s, 1);                            
  res = s->set_bpc(s, 0);                            
  res = s->set_wpc(s, 1);                           
  res = s->set_raw_gma(s, 1);                   
  res = s->set_lenc(s, 1);                           
  res = s->set_wb_mode(s, 0);                  
  res = s->set_ae_level(s, 0);                     
  delay(1000);
}`

cocohgo, 2019-10-30:
This prints on the serial port

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8

ssid: Lab_118
password: 

Camera Ready! Use 'http://192.168.1.111' to connect
/esp32-cam
-LsSua4SA589dOA_yypB
/esp32-cam/-LsSua4SA589dOA_yypB
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0

mobizt, 2019-10-31:
It may not possible.

The library has the built-in function to read and store File and BLOB (binary) data by convert these data to base64 string and back.

But from implementation, store the large string in Firebase has some limitations. 

The data larger than 10k (which convert to bigger 15k base64 string) will be failed due to some Firebase limitations or something that undocumented. 

I already state this in [example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/File/File.ino#L120).

cocohgo, 2019-10-31:
I had seen that example but i didn't know if there was any other way to do it.
Then it is not possible to do it

closed 2019-10-31 by mobizt

mobizt, 2019-11-03:
Now the library was supported large file read and store in latest v 3.5.5.

See example here for [SD](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/File/File.ino) or [SPIFFS](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/File_SPIFFS/File_SPIFFS.ino) data.

oren-tal, 2023-04-18:
Hi, we have the same problem and the two example links above seems to be broken.
may you please refer us to the fix please?

mobizt, 2023-04-18:
@oren-tal 

You are posting on outdated topic.
Check the examples here.
https://github.com/mobizt/Firebase-ESP32/tree/master/examples