mobizt/Firebase-ESP32 #40: HELP FSR402 doesn't work with Firebase-ESP32

sarkrui, 2019-11-03:
### Specs
- Board: Wemos Lolin32 Lite
- Sensor: FSR402 (connected to GPIO 4)
- Library version: lateset (3.5.4)

const int sensorPin = 4;
int pressureVal;

### Issue
Hi @mobizt, my FSR402 works just fine without Firebase; But if I incorporate `Firebase-ESP32` into the code, the reading of the sensor pin outputs a constant value `4095` for no reason. 

### What I have tried
- commenting Firebase codes in the loop didn't solve the issue (commenting codes in the setup did though)
- MPU6050 (6DOF sensor) works well with Firebase-ESP32

Here it's one piece of my code.
>Serial.println(pressureVal);
> setInt(firebaseData, path, pressureVal); 
> //also try with 
> //pushInt(firebaseData, path, pressureVal); 

Do you have any idea where might go wrong? 
Thanks!

mobizt, 2019-11-03:
Nothing related to Firebase library.

FSR402 is actually a ceramic type piezoelectric transducer. This transducer obtains electric charge or voltage when force applied, then you can read it though analog pin or use Op-Amp to amplify the signal.

mobizt, 2019-11-03:
You should share your code for better investigation.

sarkrui, 2019-11-03:
I also thought it shouldn't be related to the Firebase library...but things just couldn't work as intended and here's the code.
[FSR402_Firebase_Send.zip](https://github.com/mobizt/Firebase-ESP32/files/3801379/FSR402_Firebase_Send.zip)

mobizt, 2019-11-03:
You cannot use **ADC2 (GPIO 2, 4, 12, 13, 14, 15, 25, 26, 2)** with WiFi task, here is the [issue](https://github.com/espressif/arduino-esp32/issues/440).

closed 2019-11-03 by mobizt

sarkrui, 2019-11-03:
_Issue resolved by changing sensor pin to ADC1._