mobizt/Firebase-ESP32 #42: How to properly monitor several streams for changes?

mrdc, 2019-11-03:
Hello,

How to properly monitor changes in several streams:
/Topic1
/Topic2
/TopicN

This way it doesn't work:
```c
if (!Firebase.beginStream(firebaseData, "/Topic1"))
{
  Serial.println(firebaseData.errorReason());
}
if (!Firebase.beginStream(firebaseData, "/Topic2"))
{
  Serial.println(firebaseData.errorReason());
}

void loop() {
...
if (firebaseData.streamAvailable())
{
if firebaseData.stringData == "Something" {
some code here
  }
}
```

Or should I do it like this:
```c
if (!Firebase.beginStream(firebaseTopic1, "/Topic1"))
{
  Serial.println(firebaseData.errorReason());
}
if (!Firebase.beginStream(firebaseTopic2, "/Topic2"))
{
  Serial.println(firebaseData.errorReason());
}

void loop() {
...
if (firebaseTopic1.streamAvailable())
{
if firebaseTopic1.stringData == "Something" {
some code here
  }
}
if (firebaseTopic2.streamAvailable())
{
if firebaseTopic2.stringData == "Something" {
some code here
  }
}

```

mobizt, 2019-11-03:
For stream, you should use one FirebaseData object for one stream path, multiple stream paths for one FirebaseData object is not possible.

Use multiple FirebaseData objects consume more memory and lead to out of memory.

**Firebase.beginStream** should be in setup() for each stream path per a FirebaseData object.

For stream without callback function, in loop() you need to call **Firebase.readStream** before **Firebase.streamAvailable**, see [this example](mobizt/-ESP32/blob/Firebasemaster/examples/Stream_loop/Stream_loop.ino).

For stream with callback, you're not need to read and check stream manually, just assign the callbak function in setup(), see [this example](
https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Stream_callback/Stream_callback.ino).

For others Firebase calls e.g. get, set, push, update, and delete, the FirebaseData object should be used separately from stream because stream connection need to be stay alive and not interrupt by these calls.

One FirebaseData object can be shared for use among Firebase calls (get, set, push, update and delete).

closed 2019-11-04 by mobizt

mrdc, 2019-11-04:
@mobizt 
Thanks!

> Use multiple FirebaseData objects is not a good practice because they consume more memory and leads to out of memory in your device. You should have design plan to select the appropiate and true necessary stream paths.

That's pity. Looks like that I need to include several parameters in one Stream, to save memory and not making multiple Firebase objects: Topic: "Parameter1; Parameter2; ParameterN".
What about modules with PSRAM? Does your library able to use it by default?

> **Firebase.beginStream** should be in setup() for each stream path per a FirebaseData object.

Yes, that's exactly the same what I've meant in the second part of my post.

> For stream without callback function, in loop() you need to call **Firebase.readStream** before **Firebase.streamAvailable**, see [this example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Stream_loop/Stream_loop.ino).

I have it in my code - haven't included it here.

mobizt, 2019-11-04:
PSRAM is not support.
I don't mean all your stream data should be in the same parent node unless your should compromise between the use of FirebaseData object and your free heap memory.