mobizt/Firebase-ESP32 #44: Pull data from void function and store them to int? How?

lem0928, 2019-11-10:
I followed the basic.ino example provided. I learned how to serial print using the void function printResult(dataObj). Now how do I store the data into an int variable without serial printing it?
.
My Code:
//
FirebaseData testData;
//
String path = "/Main";
void setup()
{
Serial.begin(9600);
Firebase.setInt(testData, path, 0);
}
//
void printResult(FirebaseData &data)
{
if (data.dataType() == "int")
    Serial.println(data.intData());
  else if (data.dataType() == "float")
    Serial.println(data.floatData(), 5);
  else if (data.dataType() == "double")
    printf("%.9lf\n", data.doubleData());
  else if (data.dataType() == "boolean")
    Serial.println(data.boolData() == 1 ? "true" : "false");
  else if (data.dataType() == "string")
    Serial.println(data.stringData());
}
//
Firebase.getInt(testData, path);
//
int data1 = // here is where I want to store the int value data.intData() pulled from the printResult function without serial printing the value, how? 
//

mobizt, 2019-11-10:
You can get the server returned data via FirebaseData object, which is testData for your case.

**int data1 = testData.intData();**
Or check the type of data with
**String dataType = testData.dataType();**

lem0928, 2019-11-10:
Sir, for the past week, I've been a huge pestering guy for you. Now I think I've learned enough for my projects. It's all thank to you and the library you've provided. Thank you!

closed 2019-11-11 by mobizt