mobizt/Firebase-ESP32 #46: HELP:returning  json data despite of  posting bool data from hardware.

vamsikrishnaA9, 2019-11-18:
hi,
i am posting bool data from my hardware to data base. each time strem data function triggered and returning me the json data instead of bool data.
here is my working code.

#include <WiFi.h>
#include <FirebaseESP32.h>
#define WIFI_SSID "xxxxxxx"
#define WIFI_PASSWORD "xxxxxxx"
#define FIREBASE_HOST "xxxxxxxxx"
#define FIREBASE_AUTH "xxxxxxxxx"
const int relayPin1 = 19;
//Define Firebase data object
FirebaseData firebaseData;
FirebaseJsonData JsonData;
bool status1 = false;
String path = "xxxxx/xxx/xx/xx/xx";
int temp[]={1};
void printResult(FirebaseData &data);

void setup()
{
  Serial.begin(115200);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void loop()
{
  if (Firebase.readStream(firebaseData) && firebaseData.streamAvailable())
  {
    Serial.println("------------------------------------");
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData.streamPath());
    Serial.println("EVENT PATH: " + firebaseData.dataPath());
    Serial.println("DATA TYPE: " + firebaseData.dataType());
    Serial.println("EVENT TYPE: " + firebaseData.eventType());
    Serial.print("VALUE: ");
    printResult(firebaseData);
   // Serial.println("------------------------------------");
    Serial.println();
        if (digitalRead(4)==HIGH)
  {
  Firebase.setBool(firebaseData,path+"/switch_1/status",true);
  }
  else{
   Firebase.setBool(firebaseData,path+"/switch_1/status",false); 
  }
  }
}
void printResult(FirebaseData &data)
{
  if (data.dataPath() == "/switch_1/status" && data.dataType() == "boolean") {
    temp[0] = data.boolData(); digitalWrite(relayPin1, !data.boolData()); Serial.println(data.boolData());
  }
    else if (data.dataType() == "json")
  {   
      Serial.println();
      FirebaseJson &json = data.jsonObject();
      json.get(JsonData, "/switch_1/status");
      temp[0] = JsonData.boolValue;
      digitalWrite(relayPin1, !temp[0]);
      Serial.print(temp[0]);          
  }
}

when i am changing the firebase data from firebase console it was returning bool data which is correct.
problem with  when i triggered the data from my hardware.
thank you. please suggest me the needful.

mobizt, 2019-11-18:
This is because you set the stream path at some path and set the data at its child path.

The return payload data from server is **JSON** as the following.

**{"switch_1":{"status":true}}** and
**{"switch_1":{"status":false}}**

That is child value of your stream path at "xxxxx/xxx/xx/xx/xx"

If stream path is "xxxxx/xxx/xx/xx/xx/switch_1/status" then you will get bool data.

mobizt, 2019-11-18:
Then you need to parse JSON data belong to "xxxxx/xxx/xx/xx/xx" to get the child value as you want.

This is the basic of JSON object and Firebase which you need to know.

closed 2019-11-18 by mobizt

vamsikrishnaA9, 2019-11-18:
#include <WiFi.h>
#include <FirebaseESP32.h>
#define WIFI_SSID "SPVF"
#define WIFI_PASSWORD "spvf1809"
#define FIREBASE_HOST "xxxxxxxxxxxxxxxxxxxxxxxxx"
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxx"
const int relayPin1 = 19;
const int relayPin2 = 33;
//Define Firebase data object
FirebaseData firebaseData;
FirebaseJsonData JsonData;
String path = "/smartJoules/vamsikrishnavuyyuru007/devices/Hall/connections";
int temp[] = {1};
void printResult(FirebaseData &data);

void setup()
{
  Serial.begin(115200);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}
void loop()
{
  if (Firebase.readStream(firebaseData) && firebaseData.streamAvailable())
  {
    Serial.println("------------------------------------");
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData.streamPath());
    Serial.println("EVENT PATH: " + firebaseData.dataPath());
    Serial.println("DATA TYPE: " + firebaseData.dataType());
    Serial.println("EVENT TYPE: " + firebaseData.eventType());
    Serial.print("VALUE: ");

    printResult(firebaseData);
    Serial.println("------------------------------------");
    Serial.println();
    if (digitalRead(4) == HIGH)
    {
      Firebase.setBool(firebaseData, path + "/switch_1/status", true);
    }
    else if(digitalRead(4) == LOW)
    {
      Firebase.setBool(firebaseData, path + "/switch_1/status", false);
    }
        if (digitalRead(5) == HIGH)
    {
      Firebase.setBool(firebaseData, path + "/switch_2/status", true);
    }
    else if(digitalRead(5) == LOW)
    {
      Firebase.setBool(firebaseData, path + "/switch_2/status", false);
    }
  }

}
void printResult(FirebaseData &data)
{
  if (data.dataPath() == "/switch_1/status" && data.dataType() == "boolean") 
  {
    temp[0] = data.boolData(); digitalWrite(relayPin1, !data.boolData());
    Serial.println(data.boolData());
    Serial.println("data 1");
    Serial.println(temp[0]);
  }
  if (data.dataPath() == "/switch_2/status" && data.dataType() == "boolean") 
  {
    temp[1] = data.boolData(); digitalWrite(relayPin2, !data.boolData());
    Serial.println(data.boolData());
    Serial.println("data 2");
    Serial.println(temp[1]);
  }
  else if (data.dataType() == "json")
  {

    Serial.println();
    FirebaseJson &json = data.jsonObject();
    json.get(JsonData, "/switch_1/status");
    temp[0] = JsonData.boolValue;
    digitalWrite(relayPin1, !temp[0]);
       Serial.print(temp[0]);
    json.get(JsonData, "switch_2/status");
    temp[1] = JsonData.boolValue;
    digitalWrite(relayPin2, !temp[1]);
    Serial.print(temp[1]);
   
  }
}

This is my code. What i am doing is using 2 relays and 2 push buttons for triggering them. I have created a tree structure in firebase which i have attached screen shot . Initially my esp32 will connect to the wifi and then the database and reads all the states for the 1st time in the database tree and retrieve in the json structure so that i will write those states to both my relays(I HAVE ATTACHED SCREEN SHOT OF THE RESPONSE). Now the relays states are set and waiting for change of state. When i change the state in the database tree the response i am getting is boolean because i am hitting as boolean, but when i push a push button and i post the new  button state using setbool() to update the relay state i am getting stream response as Json(I HAVE ATTACHED SCREEN SHOT OF THE RESPONSE) which gives me the data of both the relays instead of the relay state or switch_1 state i posted from esp32. 
my requirement is 
1. I need both the relays states for the 1st time and write to both relays
2. i need to post the new relay state when the push button is pressed and need to get only the corresponding relay data as boolean because im posting boolean.

[image]
[image]
[image]

vamsikrishnaA9, 2019-11-18:
If i had written any mistake in the code please elaborate and explain what went wrong because i am somewhat new to this.

mobizt, 2019-11-18:
That is normal and depends on Firebase server. 

Firebase web console is not a real client, it is used for manage the data.

The result of REST client (this library) will be as you get from library.

I can't do anything with server side operation.

The Firebase (Google) server returns payload data as client send a request which will be JSON, string, Boolean, null and number.

Library determines the types of payload from these situation (condition)

JSON which always starts with **{** and ends with **}**
String which always starts and end with quote (**"**)
Boolean which will be **true** or **false**
null will be **null**
Number which not match the above data. 

Integer is the numbers that not contained period (**.**) otherwise it will be float or double. 
If its double value is less than 0x7fffffff, it will be determined as float .

vamsikrishnaA9, 2019-11-18:
posting from firebase console mean i am posting boolean data from an android application to the database. Android application can be considered as a real client right ?
while posting boolean from the android app to database the stream response is coming out as boolean only, but problem persists while posting from ESP32 module only despite posting boolean,int,string etc data using set() functions. Iam not able to understand why?

mobizt, 2019-11-18:
Your code leads to some errors. 

1. Because of use of the same Firebase Data object, your stream will pause a while when you call oher Firebase function which share the same Firebase Data object (WiFi client).

2. If you use the different Firebase Data object for stream and other Firebase calls, with the current code, the data collision may be occurred, due to it shared the same WiFi rx buffer. Server will return unordered response when you set the change immediately to the node or path that being stream when stream event (changes) occurred.

The second error will be fixed in the next update.

vamsikrishnaA9, 2019-11-18:
Thank you soo much for your quick response. Awaiting next update :)

mobizt, 2019-11-18:
If you want someone to help, clear information is important.

vamsikrishnaA9, 2019-11-18:
was the information posted by me sufficient or you want to know anything more specific ?

mobizt, 2019-11-18:
Ok due to you code make me a litle bit confused.
I remove my post above after clearify the problem.
Actually the library was OK unless you need to use another Firebase Data Object for set function.

This because when you use the same Firebase Data Object for stream and set function, when you call set, everything belong to stream will be reset and after that it switch to stream mode with **Firebase.readStream** then it returns the whole change (all JSON) at stream path.

```C++

#include <WiFi.h>
#include <FirebaseESP32.h>
#define WIFI_SSID "XXXXXXX"
#define WIFI_PASSWORD "XXXXXXX"
#define FIREBASE_HOST "xxxxxxxxxxxxxxxxxxxxxxxxx"
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxx"

const int relayPin1 = 19;
const int relayPin2 = 33;
//Define Firebase data object
FirebaseData firebaseData;
FirebaseData firebaseData2;
FirebaseJsonData JsonData;
String path = "/smartJoules/vamsikrishnavuyyuru007/devices/Hall/connections";
int temp[] = {1, 1};
void printResult(FirebaseData &data);
void setup()
{
  Serial.begin(115200);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}
void loop()
{
  if (Firebase.readStream(firebaseData) && firebaseData.streamAvailable())
  {
    Serial.println("------------------------------------");
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData.streamPath());
    Serial.println("EVENT PATH: " + firebaseData.dataPath());
    Serial.println("DATA TYPE: " + firebaseData.dataType());
    Serial.println("EVENT TYPE: " + firebaseData.eventType());
    Serial.print("VALUE: ");
    printResult(firebaseData);
    Serial.println("------------------------------------");
    Serial.println();
    if (digitalRead(4) == HIGH)
    {
      Firebase.setBool(firebaseData2, path + "/switch_1/status", true);
    }
    else if (digitalRead(4) == LOW)
    {
      Firebase.setBool(firebaseData2, path + "/switch_1/status", false);
    }
    if (digitalRead(5) == HIGH)
    {
      Firebase.setBool(firebaseData2, path + "/switch_2/status", true);
    }
    else if (digitalRead(5) == LOW)
    {
      Firebase.setBool(firebaseData2, path + "/switch_2/status", false);
    }
  }
}
void printResult(FirebaseData &data)
{
  if (data.dataPath() == "/switch_1/status" && data.dataType() == "boolean")
  {
    temp[0] = data.boolData();
    digitalWrite(relayPin1, !data.boolData());
    Serial.println(data.boolData());
    Serial.println("data 1");
    Serial.println(temp[0]);
  }
  if (data.dataPath() == "/switch_2/status" && data.dataType() == "boolean")
  {
    temp[1] = data.boolData();
    digitalWrite(relayPin2, !data.boolData());
    Serial.println(data.boolData());
    Serial.println("data 2");
    Serial.println(temp[1]);
  }
  else if (data.dataType() == "json")
  {
    Serial.println();
    FirebaseJson &json = data.jsonObject();
    json.get(JsonData, "/switch_1/status");
    temp[0] = JsonData.boolValue;
    digitalWrite(relayPin1, !temp[0]);
    Serial.print(temp[0]);
    json.get(JsonData, "switch_2/status");
    temp[1] = JsonData.boolValue;
    digitalWrite(relayPin2, !temp[1]);
    Serial.print(temp[1]);
  }
}
```

vamsikrishnaA9, 2019-11-19:
Thanks mate, understood the issue now and sorted out the problem. Thanks you soo much for quick response and resolution.