mobizt/Firebase-ESP32 #47: ENHANCEMENT

gibus27, 2019-11-26:
This is not really a problem but a request: do you think it is possible to implement the use of the FireBase Storage (**not** FireBase DataBase)?
Example: store a picture as a file.
Thank you,
Alain

mobizt, 2019-11-26:
It will be available in the next update which not published in the near future due to it takes time to develop.
The next update included many features and changes, here is the progress.

Support OAuth2.0 token using Service Account... Completed
Cloud Storage... Completed
Cloud Firestore through gRPC... Under development.

gibus27, 2019-11-27:
Thank you very much for your answer.
It's really a very good development.
Thanks again

closed 2019-11-27 by gibus27

99bit, 2020-05-17:
@mobizt when this features will be realise??!(firestore support )

mobizt, 2020-05-17:
The realtime feature of Firestore is not supported via REST APIs, then it's not worth to work in this way.

I am going to use protobuf (Google protocol buffer) and gRPC framework through HTTP 2 which recommended by google but it'ss not available (can't find the open-source library that targets to ESP32 right now.

99bit, 2020-05-17:
@mobizt 

Thank you for quick response

And also for gr8 library it made my word way easy 😍😍