mobizt/Firebase-ESP32 #48: HELP

fervenceslau, 2019-11-28:
I updated the library to the most recent version and my code stopped working. 

I was using Firebase.updateNode(firebaseData, "\", json) to upload a single json structure to firebase, but now I get an error.

After this update, I feel that the documentation is very confusing and it is making a lot harder to figure out what to do. For example, the "Changes from earlier version" section states that "getJson, setJson, pushJson, updateNode and updateNodeSilent." have been changed, but does this mean they are not supported anymore? Also, there are a lot of functions with similar names such as "setJsonData" and "setJson", but I dont know what's the difference.

Finally, you could add a content table for the functions descriptios to make it easier to find and understand their usage.

Obs.: Great library!

closed 2019-11-28 by fervenceslau

mobizt, 2019-11-28:
For new version, the JSON string was not supported by the functions that take String of serialized JSON object e.g. setJson and updateNode.

You need to use FirebaseJson object (class) for that functions as they accept FirebaseJson object as parameter.

To set the JSON string (serialized string) to FirebaseJson, you can call setJsonData from FirebaseJson object.

The examples and the document (bottom section of doc) are also updated for the FirebaseJson usages.

mobizt, 2019-11-29:
The updateNode functions accept only **FirebaseJson** object
```C++
bool updateNode(FirebaseData &dataObj, const String &path, FirebaseJson &json);
bool updateNodeSilent(FirebaseData &dataObj, const String &path, FirebaseJson &json);
bool updateNodeSilent(FirebaseData &dataObj, const String &path, FirebaseJson &json, float priority);
```

You can set the JSON string like this.
```C++
FirebaseJson json;
json.setJsonData ("{\"key\":\"value\"}");
```
or 
```C++
FirebaseJson json;
String jsonStr = "{\"key\":\"value\"}";
json.setJsonData (jsonStr);
```
Or create it from add and set functions.
```C++
FirebaseJson json;
json.add ("key", "value");
json.add ("myInt", 123);
```
or
```C++
json.set("/key", 99.25);
json.set("/parent/child", "test");
```

Then pass json to updateNode
```C++
updateNode(firebaseData,  "path to your node", json);
```

fervenceslau, 2019-11-29:
Got it to work now, thanks! 

By the way, is there a limit to how much data you can send at once? I tried updating a node with more than 512 double values and it didn't work. I had to split the data into three different batches to get it to work...

mobizt, 2019-11-29:
It's not limited for numbers of children nodes unless if the FirebaseJson holds too large data, free memory may not enough for WiFi Client to start SSL connection. 

For nested data, Firebase is limited to [32 levels of data depth](https://firebase.google.com/docs/database/ios/structure-data#avoid_nesting_data).