mobizt/Firebase-ESP32 #50: BUG

fervenceslau, 2019-12-18:
**Describe the bug**
Error when trying to send double data to firebase through a json object: "bad request, Invalid data; couldn't parse JSON object. Are you sending a JSON object with valid key names?".

This happens when I try to send a double value of 0.

Printing the json object I see that the sensor data (double) is being displayed as "0." when this error occurs. Whenever I set some other value to this sensor different than 0, e.g. 0.1, everything works fine...

**To Reproduce**
FirebaseJson json;
double doubleVariable = 0;

json.set(pathVariable, doubleVariable);

if (!Firebase.updateNode(gFirebaseData, path, json)) {
    Serial.println(gFirebaseData.errorReason());
    return false;
}

**IDE and its version:**
 - Arduino
 - 1.8.10

**ESP32 Arduino Core SDK version**
 - 1.0.4

**FirebaseESP32 version**
- 3.5.9

**FirebaseJson version**
- 2.2.7

mobizt, 2019-12-19:
Thanks for report this.

The bug is in double to string conversion of FirebaseJson library.

I will fix it and update soon.

mobizt, 2019-12-19:
Now the library was update with bug fixed.

closed 2019-12-20 by mobizt