mobizt/Firebase-ESP32 #51: BUG getJSON when is string

baudelioandalon, 2019-12-19:
**Describe the bug**
when I have
Test
    |__node1
            |____name:"Baudelio"
            |____money: 50

and I get the json this FirebaseData is blank is not null, is BLANK, and when I try get the reason error too is blank

**To Reproduce**
Steps to reproduce the behavior:
1. Create a node with 2 elements, one with type string and the other type int
2. Use the example jsonObject with path Test/node1
3. And is blank

**Expected behavior**
I consider that is just a error when I get different types of a json and when I received string in a json

**Screenshots**
[image]

**IDE and its version:**
 - Arduino
 - Version  1.8.10

**ESP32 Arduino Core SDK version**
 - Version LAST 19 December

**Additional context**
and when I try get the database complete don't works when I try with the path "/"

Please,I would appreciate it if you could solve it
this is my email baudelio_andalon@hotmail.com
if you want to talk I will available

mobizt, 2019-12-19:
Sorry I don't understand. 
Please share your test code and describe your code and what's your expectation.

baudelioandalon, 2019-12-19:
Sure, this is my code

#include <Arduino.h>
#include <SPI.h>
#include <MFRC522.h>
#include <WiFi.h>
#include <FirebaseESP32.h>

#define FIREBASE_HOST "rechargerrfid.firebaseio.com" //Do not include https:// in FIREBASE_HOST
#define FIREBASE_AUTH "2awqblaNgE5AyALdjempsfJa46qc9mdkLfgOZAO7"
#define WIFI_SSID "uGuest"
#define WIFI_PASSWORD "UberConnect$"

#define SS_PIN 5
#define RST_PIN 22

String path = "/Test/tarjetas";
FirebaseData firebaseData;
int nointernet = 0;
void printResult(FirebaseData &data);
 
MFRC522 rfid(SS_PIN, RST_PIN); // Instance of the class
MFRC522::MIFARE_Key key; 

boolean readed = false;
String codigo = "";

void setup() { 
  Serial.begin(115200);
  
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Conectando a Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
    nointernet++;
    if(nointernet>4){
      Serial.println("Reiniciando");
      delay(700);
      esp_restart();
    }
  }

  Serial.print("Conectado IP: ");
  Serial.println(WiFi.localIP());
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);

  SPI.begin(); // Init SPI bus
  rfid.PCD_Init(); // Init MFRC522 

}
 
void loop() {

  // Reset the loop if no new card present on the sensor/reader. This saves the entire process when idle.
  if ( rfid.PICC_IsNewCardPresent()){
      // Verify if the NUID has been readed
    if ( rfid.PICC_ReadCardSerial()){
        
        MFRC522::PICC_Type piccType = rfid.PICC_GetType(rfid.uid.sak);
        // Check is the PICC of Classic MIFARE type
          if (piccType != MFRC522::PICC_TYPE_MIFARE_MINI &&  
              piccType != MFRC522::PICC_TYPE_MIFARE_1K &&
              piccType != MFRC522::PICC_TYPE_MIFARE_4K) {
              Serial.println("Esta tarjeta no es aceptada");
              return;
          }else{
              
              printHex(rfid.uid.uidByte, rfid.uid.size);             
              rfid.PICC_HaltA();
              rfid.PCD_StopCrypto1();
          }
    }

  }

}//loop

void printHex(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {

    if(buffer[i] < 0x10){
      codigo = codigo + "0";
    }
    codigo = codigo + String(buffer[i], HEX);
   
  }
  consultar(codigo);
  codigo = "";

}

//Here I try get just name, but the result is blank and the result must don't be blank
void consultar(String uid){
      //Also can use Firebase.get instead of Firebase.setInt
    if (Firebase.get(firebaseData, path + "/" + uid))
    {
      Serial.println("Buscando el uid: " + uid);
      if(firebaseData.dataType() == "null"){
        Serial.println("No existe");
      }else{
        String jsonStr = "";
        FirebaseJson json1; 
        FirebaseJsonData jsonObj;
        json1.toString(jsonStr, true);
        jsonStr = firebaseData.jsonString();
        json1.setJsonData(jsonStr);
        json1.get(jsonObj, "name");
        String saldo = jsonObj.stringValue;
        Serial.println(saldo);
            
      }

    }
}

void printResult(FirebaseData &data)
{

   if (data.dataType() == "json"){
    Serial.println();
    FirebaseJson &json = data.jsonObject();

    String jsonStr;
    json.toString(jsonStr, true);
    Serial.println(jsonStr);
    Serial.println("Iterate JSON data:");
    size_t len = json.iteratorBegin();
    String key, value = "";
    int type = 0;
    for (size_t i = 0; i < len; i++)
    {
      json.iteratorGet(i, type, key, value);
      Serial.print(i);
      Serial.print(", ");
      Serial.print("Type: ");
      Serial.print(type == JSON_OBJECT ? "object" : "array");
      if (type == JSON_OBJECT)
      {
        Serial.print(", Key: ");
        Serial.print(key);
      }
      Serial.print(", Value: ");
      Serial.println(value);
    }
    json.iteratorEnd();
  }else if(data.dataType() == "string"){
    Serial.println(data.stringData());
  }else if (data.dataType() == "array")
  {
    Serial.println();
    //get array data from FirebaseData using FirebaseJsonArray object
    FirebaseJsonArray &arr = data.jsonArray();
    //Print all array values
    Serial.println("Pretty printed Array:");
    String arrStr;
    arr.toString(arrStr, true);
    Serial.println(arrStr);
    Serial.println();
    Serial.println("Iterate array values:");
    Serial.println();
    for (size_t i = 0; i < arr.size(); i++)
    {
      Serial.print(i);
      Serial.print(", Value: ");

      FirebaseJsonData &jsonData = data.jsonData();
      //Get the result data from FirebaseJsonArray object
      arr.get(jsonData, i);
      if (jsonData.typeNum == JSON_BOOL)
        Serial.println(jsonData.boolValue ? "true" : "false");
      else if (jsonData.typeNum == JSON_INT)
        Serial.println(jsonData.intValue);
      else if (jsonData.typeNum == JSON_DOUBLE)
        printf("%.9lf\n", jsonData.doubleValue);
      else if (jsonData.typeNum == JSON_STRING ||
               jsonData.typeNum == JSON_NULL ||
               jsonData.typeNum == JSON_OBJECT ||
               jsonData.typeNum == JSON_ARRAY)
        Serial.println(jsonData.stringValue);
    }
  }

}

[image]

baudelioandalon, 2019-12-19:
Just I can't get a simple string and I think is an library problem, sorry if I'm wrong

mobizt, 2019-12-20:
This issue was fixed already in the latest update.

Please update the library.

closed 2019-12-20 by mobizt