mobizt/Firebase-ESP32 #52: HELP : After 2,3 hrs of connection to firebase esp32 is going unresponsive state.

vamsikrishnaA9, 2019-12-29:
Hi there, I've connected esp32 to firebase, after some moments there is no response from the module when I change the database values. 
More Info : 
I've created a task for checking internet for every 5 seconds on other core. I don't know what I have done worng got confused. Please give me suggestion that would be a great help for me.
Thank you

mobizt, 2019-12-29:
I can't hep without your test code.
Please post you working code here.

vamsikrishnaA9, 2019-12-29:
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"
#include <ESP32WebServer.h>
#undef PROVA_
#include  "ping.h"
#include <WiFiClient.h>
#include <WiFi.h>
#include <FirebaseESP32.h>
#include <RBDdimmer.h>
#include <EEPROM.h>
#define FIREBASE_HOST "smartjoules-f1445.firebaseio.com" //Do not include https:// in FIREBASE_HOST
#define FIREBASE_AUTH "8JbndeUj44B8LcwL07qynm2z2yuttnxcq54rcT1L"
#define WIFI_SSID "DIGISOL"
#define WIFI_PASSWORD "password22"

//Define Firebase data object
FirebaseData firebaseData;
FirebaseJsonData JsonData;

#define outputPin  23 // dimming pins
#define zerocross  21// for boards with CHANGEBLE input pins
//#define outputPin1 26
#define resetpin 35
#define switch0 13
#define switch1 12
#define switch2 14
#define switch3 26
#define dimming_led 18
#define red 2
#define green 4
#define blue 15

TaskHandle_t Task1;

int ia[4] = { 216, 58, 198, 164 };

//String path = "smartJoules/mukhesh2305/connections";
dimmerLamp dimmer_switch1(outputPin, zerocross);

int temp[6] = {0};
int dimVal, dim_Val, prev_dim = 20;
int var[6];
int p;
int outValue, outVal;
int speedval = 0, speed_value = 0;
float setval;
String field;

void printResult(FirebaseData &data);

void setup()
{

  Serial.begin(115200);
  EEPROM.begin(512);
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
  pinMode(switch0, OUTPUT);
  pinMode(switch1, OUTPUT);
  pinMode(switch2, OUTPUT);
  pinMode(switch3, OUTPUT);
  pinMode(dimming_led, OUTPUT);
  digitalWrite(switch0, LOW);
  digitalWrite(switch1, LOW);
  digitalWrite(switch2, LOW);
  digitalWrite(switch3, LOW);
  digitalWrite(dimming_led, LOW);
  // pinMode(19,OUTPUT);
  pinMode(blue, OUTPUT);
  pinMode(red, OUTPUT);
  pinMode(green, OUTPUT);                    //common cathode configuration
[... 171 lines cut ...]
      outVal = i;
      dimmer_switch1.setPower(outVal); // setPower(0-100%);
      delay(30);
    }
  }
  else if (buf != 0 && prev_dim > buf) {
    for (int i = prev_dim; i >= buf; i--) {
      outVal = i;
      dimmer_switch1.setPower(outVal); // setPower(0-100%);
      delay(30);
    }
  }
  prev_dim = buf;
  Serial.print(dimmer_switch1.getPower());
  Serial.println("%");
  delay(150);
}
void print_byte(byte val)  ///----------this is for serial printing the 8 bit data-------//
{
  byte i;
  for (byte i = 0; i < 8; i++)
  {
    Serial.print(val >> i & 1, BIN);
  }
}

mobizt, 2019-12-29:
Please explain more detail.
Which line get error?

vamsikrishnaA9, 2019-12-29:
Thank you for your quick response
Code is working fine. For the first 1 or 2 hrs after power up the module I am getting the stream response from the database. If I kept the module in Idle state for more than 1 or 2 hrs with out changing the values in firebase, the esp32 is not responding. Every time I need to reset the esp32 to get connection from firebase and to get values. I've used stream loop example to develop my requirement. In the second core I put continuos checking of ping to display internet status. I don't think so It creates unresponsive state problem.Please find me solution if I did anything wrong , that would be a great help for me.
Thank you very much.

mobizt, 2019-12-29:
You need to debug for free heap, memory leak and WiFi status and stream timeout checking is already available as in example.

You need more than 100k free mem for WiFi client to work.

vamsikrishnaA9, 2019-12-30:
Will work on that. Update the status to you sir. Thank you very much.

vamsikrishnaA9, 2019-12-30:
hi mobizt,
Found some observations in serial monitor,
Sketch uses 1018714 bytes (77%) of program storage space. Maximum is 1310720 bytes.
Global variables use 46128 bytes (14%) of dynamic memory, leaving 281552 bytes for local variables. Maximum is 327680 bytes.
ESP.getFreeHeap() returned max 308848 free heap memory from the void setup().
i have added ESP.getFreeHeap()  in the loop also.  the stack memory allocated to ping check task is 20k. after deducting that memory the free stack available is around 297k which keeps on depleting over time to 295k, 290k,285k........ so onn tto 200k and below. I think once the availabe memory depleting too low the module goes to unresponsive state or reset. 
This is what i have identified. I am not able to think how to proceed and resolve this memory depleting issue. any suggestions?

mobizt, 2019-12-30:
I'm not mention the stack here, only memory leak was suspected. 
The error due to stack overflown causes device to reset.

For your case, free heap available may be too low (<100k) for mbedTLS library (used by WiFi client) to operate correctly.

You need to config ArduinoIDE to debug WiFi to track its operation.

Compare free heap between when ping task removed and not removed.

vamsikrishnaA9, 2019-12-30:
when ping task was disabled the free heap was around 211780 and is stable and module does not go unresponsive. but when ping task is enabled the free heap goes on depleting and at around 65000 the module goes unresponsive because the available 65k stack is not sufficient for Wifi client to operate. This is the problem i have identified.So now how to enable ping task without depletion of free heap and what causes my free heap to deplete? how can i solve this?

mobizt, 2019-12-30:
Please try other ping libraries instead. 
I don't have any recommendation on this.

vamsikrishnaA9, 2019-12-30:
Thanks for your quick response and support. I will try to figure it out and post here if i find any solution.

closed 2019-12-30 by vamsikrishnaA9

vamsikrishnaA9, 2020-01-01:
all the ping libraries available are depleting the heap memory. so I am putting a http request to a remote server,observing the httpresponse code to check internet. now the heap memory is stabilized and no more issue.

mobizt, 2020-01-01:
Geat and thanks for inform me.