mobizt/Firebase-ESP32 #55: streamCallback gets triggered randomly

Maverick324567, 2020-01-08:
I am having this strange issue with streamCallback function. Sometimes, even if there is no change in the Firebase RTDB, callback function gets called. Can anybody please help me out?

mobizt, 2020-01-09:
It's possible that stream was timed out,  the stream restarts then you get that response.

You may need to compare the existing data and new data before process.

Maverick324567, 2020-01-09:
> It's possible that stream was timed out, the stream restarts then you get that response.
> 
> You may need to compare the existing data and new data before process.
> 
> Set the stream timeout callback to debug.

The streamTimeoutCallback doesn't get called. Only the streamCallback is getting called randomly. May be there's a problem with the database itself.

Maverick324567, 2020-01-09:
Okay, now I have a little understanding of what exactly is happening. When I modify a particular field in firebase RTDB, streamCallback gets called. Now inside streamCallback function I set a particular field in database. This is when the streamCallback gets called. I have used different FirebaseData objects for streamCallback and for setting database fields. But still I am having this issue. Can you please help me out?

mobizt, 2020-01-09:
You should to avoid to change data in stream path inside the stream callback.

If you intend to do this, you should check the returned event path. 

The event path returns from firebase is the relative path from stream path, for example

If you set the stream path to **root/myData**, and if you set the data to **root/myData/Sensor1**.

The event path returns when stream event changes will be **/Sensor1**. Then you can quit stream callback function or continue to run.

closed 2020-01-10 by mobizt