mobizt/Firebase-ESP32 #57: Coonection Refused

Shahzainali, 2020-01-17:
Hi, here are logs, I am simple writing data to realtime, Streaming data works perfectly,, I faced this issue suddenly, can you please help.

Thanks 

Connected with IP: 192.168.1.105

[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 216348
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:216] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:225] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:240] start_ssl_client(): Free internal heap after TLS 175132
[V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 171060
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:216] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:225] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:240] start_ssl_client(): Free internal heap after TLS 131592
[V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 127516
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:216] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:225] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:240] start_ssl_client(): Free internal heap after TLS 88044
[V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 83912
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():190]: (-32512) SSL - Memory allocation failed
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -32512
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 85444
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():190]: (-32512) SSL - Memory allocation failed
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -32512
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 84940
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():190]: (-32512) SSL - Memory allocation failed
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -32512
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 83740
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():190]: (-32512) SSL - Memory allocation failed
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -32512
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 83428
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():190]: (-32512) SSL - Memory allocation failed
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -32512
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 83112
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():190]: (-32512) SSL - Memory allocation failed
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -32512
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
FAILED
REASON: connection refused
------------------------------------

mobizt, 2020-01-17:
> [V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 83740

Free heap is not enough for mbedTLS to work. It requires > 100k before start SSL handshake.

You may need to reduce ram usage of your code.

Shahzainali, 2020-01-17:
Hi, I have tried but no success can you please help me here is my code..

`#include <WiFi.h>
//#include "types.h"
#include <FirebaseESP32.h>
#include <HardwareSerial.h>

#include <TinyGPS++.h>"
TinyGPSPlus gps;

bool statusL1 = true;
bool statusL2 = true;
bool statusL3 = true;

bool faultL1 = false;

HardwareSerial SerialGPS(1);

#define LIGHT1 13 
#define L1INPUT 34

#define LIGHT2 12 
#define L2INPUT 35

#define LIGHT3 14 
#define L3INPUT 32

#define LDRINPUT 33
#define AMPEREINPUT 39

#define L1INPUTF 36

// setting PWM properties

const int ledChannel1 = 0;
const int ledChannel2 = 2;;
const int ledChannel3 = 3;
const int freq = 5000;
const int resolution = 10;

//Define FirebaseESP32 data object

FirebaseData firebaseData;
FirebaseData firebaseDataCallback1;
FirebaseData firebaseDataCallback2;
FirebaseData firebaseDataCallback3;

#define RXD2 16
#define TXD2 17

void setup()
{
  Serial.begin(115200);
  SerialGPS.begin(9600, SERIAL_8N1, 16, 17);
  Serial.println("Serial Txd is on pin: " + String(TX));
  Serial.println("Serial Rxd is on pin: " + String(RX));
Serial.println(true);
  // We start by connecting to a WiFi network
  WiFi.begin("xxxx", "xxxx");
  Serial.print("Connecting to Wi - Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();
  Firebase.begin("xxxx, "xxxx");
[... 271 lines cut ...]
{
  ledcWrite(ledChannel3,32);

}
}
else{
  
  ledcWrite(ledChannel3,0);
  }
  }else{
    
  ledcWrite(ledChannel1,0);
  ledcWrite(ledChannel2,0);
  ledcWrite(ledChannel3,0);
    
    }

  int amp =  analogRead(AMPEREINPUT);
  //Serial.print(amp);
  delay(200);

  
  //pushFirebase(" / l1 / isfault", "true);
}
`

Shahzainali, 2020-01-17:
This is method for saving data `
pushFirebase("/shahzain/isfault/", true);`

previously it was working perfectly.

mobizt, 2020-01-17:
You are using too many FirebaseData objects.

Each FirebaseData object has WiFi Client instance which it consumes large amount of ram due to memory used by SSL/TLS provider ( mbedTLS library).

mobizt, 2020-01-17:
You can use only one FirebaseData object for parent node g.g. **Firebase.beginStream(firebaseDataCallback1, "/lights")**

You can check **firebaseDataCallback1.dataPath()** to determine which node has changed.

Shahzainali, 2020-01-17:
Thanks for quick response, Let me try

Shahzainali, 2020-01-17:
Thanks :) problem is fixed.

closed 2020-01-17 by Shahzainali

nguyenthang621, 2022-08-16:
I need the source code, please