mobizt/Firebase-ESP32 #58: Firebase.set() also downloads data instead of just uploading

nikrath, 2020-01-27:
Hi,
I am currently using this library alongside Firebase Monitor. What I have noticed is that when I upload some data using Firebase.setString() (or just Firebase.set) there is a download connection setup as well, as can be seen from the screenshot of the Firebase Monitor:

[image]

Although there is no amount of bytes downloaded, but according to Firebase's costing parameters, simply setting up a connection to download something also adds to the cost. So, even though no data is downloaded, still it's leading to huge amount of data download rate.

How can we solve this issue?
Thanks

mobizt, 2020-01-27:
The payload response was sent to the client as default.

Firebase provides an option to suppress this payload for GET, PUT, POST and PATCH to reduce your networking data charge.

In general set (PUT) and push (POST) functions work with specific small data, the payload returned cost as much as twice the size of your data to be set or push.

Then the library only provides a silent mode for update (PATCH) instead of set and push. 

You can use function **updateNodeSilent** instead of get. For push, the payload that contains the randomly unique string of push node path is more useful and then no alternative option for the push.

mobizt, 2020-01-27:
The download limit bandwidth, 10 GB/month for the free plan is so huge for small embedded system.

nikrath, 2020-01-27:
@mobizt Thanks a lot for such a quick response! I am currently implementing your solution and will get back to you after some monitoring.

> The download limit bandwidth, 10 GB/month for the free plan is so huge for small embedded system.

The thing is, the code uploads it's status every second to Firebase database. I have to deploy this on multiple systems as part of a project and thus, wanted to optimise the usage.  Is there a better way to upload device statistics second?

mobizt, 2020-01-27:
One second interval to store some device status is quite so often and I do not recommend it. 
I think 1 to 3 minutes is enough.

nikrath, 2020-01-27:
@mobizt I now used Firebase.updateNodeSilent() instead of Firebase.set() but Firebase Usage dashboard still shows data is being downloaded. Am I doing something wrong? Isn't there a way to simply upload data without having to incur the cost of payload response?

mobizt, 2020-01-27:
It's normal in the HTTP protocol, it always returns the response header.

mobizt, 2020-01-27:
Don't be panic about this normal thing.

nikrath, 2020-01-27:
So basically, the result of using Firebase.updateNodeSilent() and Firebase.set() is same in this case? i.e uploading data would result in download costs as well, right?

nikrath, 2020-01-27:
Also, what is the solution to this then? To increase the time interval from 1 second to somewhere around 2-3 minutes?

mobizt, 2020-01-27:
No, it's not the same, with updateNodeSilent (PATCH), the server returns 204 No Content status code (header) and no any extra payload while non-silent mode, the server returns 200 HTTP OK status code and payload (the data that you set/put/push to the server).

The solution is upon your requirement, you need to test by yourself for a long period and justify. 
The extra charge is very cheap.  I did not see anybody has this issue unless his database structure is not well designed (too complex or deep nest data).

nikrath, 2020-01-27:
@mobizt The part that I am trying to edit is literally only two levels:
root
   |
   PlantWatering
       |
       count
I am uploading only 2 bytes every 10 seconds, but the Firebase Usage page shows 10s of KiloBytes of data consumed within a minute. I have no idea how that's possible, even if we consider extra header information. Any suggestions?

mobizt, 2020-01-27:
Can you send an email without the recipient's info?

How servers know how to threaten your data without any information to guide them.

The data transmission uses a header that sends among the data.
See the [wiki](https://en.wikipedia.org/wiki/Header_(computing))

closed 2020-01-27 by mobizt