mobizt/Firebase-ESP32 #59: Example: How to parse JSON with Library

mobizt, 2020-02-04:
This is the information for everyone who wants to create, modify and parse JSON with the built-in JSON maker and parser without external JSON library.

This library introduces the 3 classes i.e. **FirebaseJson**, **FirebaseJsonArray** and **FirebaseJsonData** for creating, edit and parse JSON data.

**FirebaseJson** represents the JSON object, **FirebaseJsonArray** is for the JSON array, and **FirebaseJsonData** stores the parse information when you want to parse (extract) the data from JSON object or JSON array.

Here get JSON from Firebase and parse it.

Assume that we want to parse the JSON at node **/Data/sensors**

```C++

if (Firebase.getJSON(firebaseData, "/Data/sensors")) {

  //FirebaseJson object that returned from getJson
  FirebaseJson &json = firebaseData.jsonObject();

  //temporary object to store the data from parsing
  //this will be reused over and over
  FirebaseJsonData data;

  //parse data by key or path, the result is in FirebaseJsonData object
  json.get(data, "/path/to/node");

  //or parse the complex nested atrray in nested object
  //json.get(data, "/path/to/node/[0]/[3]"); //Where [0] is index 0, [3] is index 3

  //parsing ok?
  if (data.success) {
    
    //We determine the type of value from parsing, you can get the string by using **data.stringValue** of all data types.
    if (data.typeNum == JSON_INT) // or data.type == "int"
      Serial.println(data.intValue);
    else if (data.typeNum == JSON_BOOL) //or data.type == "bool"
      Serial.println(data.boolValue);
    else if (data.typeNum == JSON_DOUBLE) //or data.type == "double"
      Serial.println(data.doubleValue);
    else if (data.typeNum == JSON_STRING ||  data.typeNum == JSON_NULL) //or data.type == "string" || data.type == "null"
      Serial.println(data.stringValue);
    else if (data.typeNum == JSON_ARRAY) //or data.type == "array"
    {

      //get the FirebaseJson array object
      FirebaseJsonArray myArr;
      data.getArray(myArr);
      for (size_t i = 0; i < myArr.size(); i++)
      {
        //get array data at index
        myArr.get(data, i);

        //or parse JSON object or JSON array in Array
        //myArr.get(data, "/any_child/another_child/another_one");

        //parsing success?
        if (data.success) {

          if (data.typeNum == JSON_INT)
            Serial.println(data.intValue);
          else if (data.typeNum == JSON_BOOL)
            Serial.println(data.boolValue);
          else if (data.typeNum == JSON_DOUBLE)
            Serial.println(data.doubleValue);
          else if (data.typeNum == JSON_STRING )
            Serial.println(data.stringValue);
          else if (data.typeNum == JSON_ARRAY)
          {
            //parse array in array
          }
          else if (data.typeNum == JSON_OBJECT)
          {
            //parse JSON object in array
          }
        }

      } else if (data.typeNum == JSON_OBJECT) //or data.type == "object"
      {
        //parse the nested JSON object here

        FirebaseJsonA json2;
        data.getJSON(json2);
        //parse it
        json2.get(data, "/another/path/to/node");
        Serial.println(data.stringValue);

      }
    }

  }
}

```

closed 2020-02-04 by mobizt

matgrioni, 2020-02-19:
Hello,

Thanks for the library. Why is this information not in the README? It seems to me a pinned issue would make sense for a common problem, rather than desired functionality and how to use an existing feature. It would make information on the library more uniform.

mobizt, 2020-02-19:
@matgrioni 
The FirebaseJson usage is in README since it included in the library.
I add this as an issue because the issue is not only the bug or problem and someone doesn't read the README before open the issue.