mobizt/Firebase-ESP32 #61: HELP

blacxcode, 2020-02-12:
How to implement remove device token after sending a message?
I used this code but not send any more for the next user.

#define TOTAL_USER 10
#define TOTAL_DEVICE 5

for (int i = 0; i < TOTAL_USER; i++) {
     firebaseFCM.fcm.begin(FIREBASE_FCM_SERVER_KEY);
     firebaseFCM.fcm.setPriority("high");
     firebaseFCM.fcm.setTimeToLive(5000);
     firebaseFCM.fcm.setDataMessage(mDataMessage);

     int index = 0;
     for (int j = index ; j < TOTAL_DEVICE; j++) {
          firebaseFCM.fcm.addDeviceToken(mDeviceId[j]);
          index++;
     }

     Firebase.sendMessage(firebaseFCM, index-1);
     firebaseFCM.fcm.removeDeviceToken(index-1);
}

mobizt, 2020-02-12:
If firebaseFCM is FirebaseData object.

To clear all device tokens.
**firebaseFCM.fcm.clearDeviceToken();**

To get the sending error.
**firebaseFCM.errorReason();**

To send a message to a recipient, you need to provide the index of device token which should not greater than the number of device tokens you 've added minus one otherwise the error will occur.

Add and remove device tokens are just adding and removing data from a dynamic array of string. 
It is simple as you use `std::vector<String>` in C++.

And make sure you send messages to existent users or devices.

blacxcode, 2020-02-12:
I've checked, the problem is that token sent from the device is invalid. 
Thanks, Bro

closed 2020-02-12 by blacxcode