mobizt/Firebase-ESP32 #62: HELP StreamData limit Esp32

opit7, 2020-03-06:
Hello, 
is there a limit on how many StreamData instances can be initalized ? Because as soon as i use more than two for a stream with callback functions the streams stop working.

```c++
// basics
  Firebase.begin(this->config.FIREBASE_ID.c_str(), this->config.FIREBASE_SECRET.c_str());
  Firebase.reconnectWiFi(true);
  Firebase.setMaxRetry(firebaseData, 3);
  Firebase.setMaxErrorQueue(firebaseData, 30);
  Firebase.enableClassicRequest(firebaseData, true);

  // check connection
  Firebase.getBool(firebaseData, "/online") ? Serial.println("[INFO] Firebase initialized") : Serial.println("[ERROR] Firebase error: " + firebaseData.errorReason());
  if (firebaseData.dataType() == "boolean")
  {
    firebaseData.boolData() ? Serial.println("[INFO] Firebase database is online") : Serial.println("[INFO] Firebase database is offline");
    Firebase.setBool(firebaseData, this->config.FIREBASE_PATH + "/online", true);
  }

  /**** setup streams ****/
  // fbOnPing
  if (!Firebase.beginStream(test, (this->config.FIREBASE_PATH + "/cmd/ping").c_str()))
    Serial.println("[Error] Stream: " + test.errorReason());
  Firebase.setStreamCallback(test, fbOnPing, fbOnPingTimeout);

  // fbOnImpulse
  if (!Firebase.beginStream(fbDOnImpulse, (this->config.FIREBASE_PATH + "/cmd/impulse").c_str()))
    Serial.println("[Error] Stream: " + fbDOnImpulse.errorReason());
  Firebase.setStreamCallback(fbDOnImpulse, fbOnImpulse, fbOnImpulseTimeout);

  // fbOnLight
  if (!Firebase.beginStream(fbDOnLight, (this->config.FIREBASE_PATH + "/cmd/light").c_str()))
    Serial.println("[Error] Stream: " + fbDOnLight.errorReason());
  Firebase.setStreamCallback(fbDOnLight, fbOnLight, fbOnLightTimeout);

  Serial.println("[INFO] setupFirebase done");
```

[image]

Is this the wrong way to initialize multiple callback streams?

What am I missing? Any help is appreciated!

mobizt, 2020-03-07:
You are using too many FirebaseData objects.

Each FirebaseData object has a WiFi Client instance that consumes a large amount of ram due to memory used by SSL/TLS provider ( mbedTLS library).

When the free heap is less than 100k, the WiFi Client can't establish the SSL connection.

You can use only one FirebaseData object for the stream which the stream path is the parent node of children nodes that you want to track the change e.g. 
`Firebase.beginStream(firebaseData2, this->config.FIREBASE_PATH + "/cmd")`

Assume that firebaseData2 used for the stream.

You can check `firebaseData2.dataPath() `to determine which child node value has changed.

If the value of many child nodes changes at the same time the path returns from firebaseData2.dataPath()` will be **/** and the stream data will be **JSON** object which you need to parse the child node to check its value.  Then compare this child node value with its old value to determine the value changes.

mobizt, 2020-03-07:
Now the library v 3.6.8 (on Github) supports multiple paths stream.

You can update and try [this example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/MultiPath_stream/MultiPath_stream.ino).

closed 2020-03-07 by mobizt