mobizt/Firebase-ESP32 #64: BUG - Firebase only works a while

alvaroaguero55, 2020-03-19:
I'm using the library on ESP32 WROOM32 DevKit, and It works but only a few seconds/minutes and after that stop getting/pushing data to the Firebase Database and stop sending Serial.print("Loop running "); seems to freeze during firebase call.

The code is: 

`
#include <WiFi.h>
#include "FirebaseESP32.h"

#define FIREBASE_HOST "-----------.firebaseio.com"
#define FIREBASE_AUTH "----------------------------------------"
#define WIFI_SSID "-------------------"
#define WIFI_PASSWORD "----------------
[image]
"

FirebaseData firebaseData;

String sID = WiFi.macAddress();

String sboton1 = sID + "/sboton1"; 

void setup()
{
  Serial.begin(115200);
  
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
    
   
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

}

void loop()
{
 
  Serial.print("Loop running ");
 
  if (Firebase.getString(firebaseData, sboton1)) {
    Serial.println("Obteniendo el valor de sboton1: ");
    Serial.println(firebaseData.stringData());
    if (firebaseData.stringData() == "on") {
    Firebase.setString(firebaseData, sboton1, "off");               
    }
   } 

  delay(1000);
    
}

`

mobizt, 2020-03-19:
No. it's not a bug.

You need to check firebaseDat.errorReason() to see what happens.

alvaroaguero55, 2020-03-19:
I updated to the latest version 1.0.4 and is working great for now, thanks!!

closed 2020-03-19 by alvaroaguero55