mobizt/Firebase-ESP32 #66: Help --------How to send uint8_t array which has const byte  values to firebase?

noonmasho, 2020-03-28:
Hi, I'm using adafruit pn532 library (https://github.com/adafruit/Adafruit-PN532) to read nfc tags. Here it is reading the unique id into a uint8_t  array as const byte values. How can we send this data into the real time  database?

Thank you. Hope you and your loved ones are safe. Thank you.

mobizt, 2020-03-28:
You can cast the uint8_t array to const char array (c string) and pass this value to setString or pushString function.

```C++
uint8_t myArr[] = {,,,,,};
Firebase.setString(firebaseData, (const char*)myArr, "Path_to_your_node");

```
When reading it back, cast it back to uint8_t for each char.

```C++
Firebase.getString(firebaseData, "Path_to_your_node");

uint8_t *myArr2 = new uint8_t[firebaseData.stringData().length()];
for (size_t i = 0; i< firebaseData.stringData().length();i++){
  myArr2[i] = (uint8_t)firebaseData.stringData().c_str()[i];
}

```

noonmasho, 2020-03-28:
Hi,  I tried it .
Firebase.setString(firebaseData, (const char*)uid, "test1");

Since the uint8_t array has byte values it is not working properly.If you see below you can see the result in firebase. 
[image]

The same value comes in arduino serial monitor when I'm using Serial.print function - **_Serial.print(uid);_**  . When I'm individually printing the values in each position using a for loop as given below it is working out fine.

```
for (uint8_t i=0; i < 7; i++) 
    {
      Serial.println(uid[i]); 
    }
```
The answer for the above code in serial monitor is this;

119
153
148
53
0
0
0

I think that the issue is in converting byte to char or string .  When the number of digits in a position is above 1 it is not working. What can we do here?

I want the result in firebase like this;

11915314853000

mobizt, 2020-03-28:
It is because the string is a null-terminated char array. If any byte data in your byte array is zero (null char), the casted string will be truncated.

Binary data in Firebase RTDB is not supported. The library provides an indirect way to store these binary data as base64 encoded string with function setBlob, pushBlob and getBlob.

Please see [this example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Blob/Blob.ino) for the usage.

noonmasho, 2020-03-30:
Hi , thank you for the fast response. I have one query.  Is there a way where we can send data without  the prefix **_blob,base64,_** . It's coming like this in the rtdb.
[image]

and I wish to have just the encoded value alone. What can be done here? 

Thanking you in advance.

mobizt, 2020-03-30:
Did you see the example?
Use getBob function.

noonmasho, 2020-03-30:
I'm making an android app where a string query will be done. If it is just the encoded string then it is easy to do in the RTDB , right?

mobizt, 2020-03-30:
See this https://github.com/mobizt/Firebase-ESP32/issues/65#issuecomment-602629455

noonmasho, 2020-03-30:
Thank you, but won't it make the app slower with more computations. I'm just worried about the computation time.

mobizt, 2020-03-30:
Do you know about base64 encode/decode algorithm?
It just the basic thing in the computer world.

It also use in email transactions. The attachment (any binary data) will be converted to base64 string.
It can implement fast in 8-bit MCU, low clock speed as MCS51 too.

Please learn how to encode and decode first.

closed 2020-03-30 by mobizt