mobizt/Firebase-ESP32 #68: BUG

abhijit-031991, 2020-04-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**IDE and its version:**
 - Arduino, PlatformIO
 - Version [e.g. 1.8.9]

**ESP8266 Arduino Core SDK version**
 - Version [e.g. 2.5.1]

**Additional context**
Add any other context about the problem here.

Hello, 

the library doesn't seem to be compiling on platformio. I have copied the error msg below.
This error is occuring even on the example files. Please take a look.

**In file included from C:/Users/LENOVO/.platformio/lib/Firebase ESP32 Client_ID6217/src/FirebaseESP32.h:49:0,
                 from src\main.cpp:18:
C:/Users/LENOVO/.platformio/lib/Firebase ESP32 Client_ID6217/src/FirebaseESP32HTTPClient.h:48:1: error: expected class-name before '{' token 
 {
 ^
C:/Users/LENOVO/.platformio/lib/Firebase ESP32 Client_ID6217/src/FirebaseESP32HTTPClient.h:106:3: error: 'TransportTraitsPtr' does not name a type
   TransportTraitsPtr transportTraits;
   ^
C:/Users/LENOVO/.platformio/lib/Firebase ESP32 Client_ID6217/src/FirebaseESP32HTTPClient.h:96:25: error: 'HTTPCLIENT_DEFAULT_TCP_TIMEOUT' was not declared in this scope
   uint16_t tcpTimeout = HTTPCLIENT_DEFAULT_TCP_TIMEOUT;
                         ^
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
======================================================== [FAILED] Took 11.60 seconds ========================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.** 

Please take a look. Please let me if my copy of vsc/platformio needs some changes.

mobizt, 2020-04-06:
The library you're using is for ESP32. 

Please install the Firebase library for ESP8266 instead.

Also please update the ESP8266 Arduino Core in pio too.

closed 2020-04-06 by mobizt

mobizt, 2020-04-07:
Sorry for the misunderstood about the library version in the previous reply.

The library was written on PlatformIO and has been tested for Arduino and PlatformIO environment. 

You need to update your PlatformIO by run this command in the console.
```
pio update
```

If you still get the problem, please uninstall and reinstall PlatformIO.

mobizt, 2020-04-07:
The library was written on PlatformIO and has been tested for Arduino and PlatformIO environment. 

You need to update your PlatformIO by run this command in the console.
```
pio update
```

If you still get the problem, please uninstall and reinstall PlatformIO.