mobizt/Firebase-ESP32 #69: Test:29: error: 'FirebaseJson' does not name a type

luisrival, 2020-04-08:
Hi I´m trying to run the basic example and when I try to compile, I can´t saying this:

do someone know if I´m doing something wrong?

`Arduino: 1.6.13 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

WARNING: library Firebase_ESP32_Client claims to run on [esp32] architecture(s) and may be incompatible with your current board which runs on [esp8266] architecture(s).
Test:27: error: 'FirebaseData' does not name a type

 FirebaseData firebaseData;

 ^

Test:29: error: 'FirebaseJson' does not name a type

 FirebaseJson json;

 ^

Test:31: error: variable or field 'printResult' declared void

 void printResult(FirebaseData &data);

                  ^

Test:31: error: 'FirebaseData' was not declared in this scope

Test:31: error: 'data' was not declared in this scope

 void printResult(FirebaseData &data);

                                ^

C:\Users\luisr\Documents\Arduino\Test\Test.ino: In function 'void setup()':

Test:50: error: 'Firebase' was not declared in this scope

   Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);

   ^

Test:54: error: 'firebaseData' was not declared in this scope

   Firebase.setReadTimeout(firebaseData, 1000 * 60);

                           ^

Test:81: error: 'printResult' was not declared in this scope

       printResult(firebaseData);

                               ^

Test:107: error: 'printResult' was not declared in this scope

       printResult(firebaseData);

                               ^

Test:151: error: 'json' was not declared in this scope

     json.clear().add("Data" + String(i + 1), i + 1);

     ^

Test:180: error: 'json' was not declared in this scope

     json.set("Data" + String(i + 1), i + 5.5);

     ^

Test:189: error: 'printResult' was not declared in this scope

       printResult(firebaseData);

                               ^

C:\Users\luisr\Documents\Arduino\Test\Test.ino: At global scope:

Test:203: error: variable or field 'printResult' declared void

 void printResult(FirebaseData &data)

                  ^

Test:203: error: 'FirebaseData' was not declared in this scope

Test:203: error: 'data' was not declared in this scope

 void printResult(FirebaseData &data)

                                ^

exit status 1
'FirebaseData' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

mobizt, 2020-04-08:
Why install a library for ESP32 for the ESP8266 device?

Pleas install this library instead https://github.com/mobizt/Firebase-ESP8266

closed 2020-04-08 by mobizt

sriharibg, 2022-07-07:
hi mobizt, I installed the library you've suggested, but still, I'm getting the same error. Is there any other solution i should try?

mobizt, 2022-07-07:
@[sriharibg](https://github.com/sriharibg)

You should open the new issue with all compilation errors because different causes.