mobizt/Firebase-ESP32 #72: [ENHANCEMENT] "Connection lost" while trying to connect over GSM/3G

jrborbars, 2020-05-22:
Hi,
I'm trying to connect to firebase over a gsm connection, using a Lilygo TTGO ESP32-SIM800 module, but getting only "connection lost" in error_reason(), even if the mobile network is connected. Reading the code, I've found in line 317 this code:
```
if (dataObj._firebaseCall && WiFi.status() != WL_CONNECTED)
        return false;
```
So, this library is only intended to be used with a wifi connection?
If not, there are a smooth way to do the connection over gsm?
if yes, maybe I can help with an abstraction of connection test.

mobizt, 2020-05-23:
Yes the library uses the WiFiClientSecure library to connect.
Any bridge connection is not supported unless LAN.

mobizt, 2020-05-23:
The bridge connection should support SSL and able to access and control the rx and tx data stream.

closed 2020-05-23 by mobizt