mobizt/Firebase-ESP32 #75: HELP

TeohWeiWei, 2020-06-06:
I have type these in my arduino code 
[...
FirebaseJson json;
FirebaseJsonArray jsonArr;

jsonArr.addInt(measure);
json.clear().addArray("data", &jsonArr).addInt("timestamp", 0); 
...]

The error that I have faced in Arduino are below: 
cardio_sender:61:13: error: 'class FirebaseJsonArray' has no member named 'addInt'
cardio_sender:65:16: error: 'class FirebaseJson' has no member named 'addArray'
cardio_sender:68:16: error: 'class FirebaseJson' has no member named 'addInt'

It works when once I have changed the first code from "jsonArr.addInt(measure);" to "jsonArr.add(measure);"
but for the second code "json.clear().addArray("data", &jsonArr).addInt("timestamp", 0); " still occurred error if I change to "json.clear().add("data", &jsonArr).add("timestamp", 0); ".

Can anyone help me to solve it :(

mobizt, 2020-06-06:
There are many functions removed and added to the FirebaseJson class when you updated from the (too) old version.

You just use **add** instead of **addXXX** 

```C++
  FirebaseJson json;
  FirebaseJsonArray jsonArr;

  jsonArr.add(measure);
  json.clear().add("data", jsonArr).add("timestamp", 0);
```

Please see the examples for the  FirebaseJson usage.

TeohWeiWei, 2020-06-06:
Hi thanks for reply.
I have try it but still have error :(
Is there any solutions to solve this problem?

mobizt, 2020-06-06:
Update the library.
Copy and post the error here.

TeohWeiWei, 2020-06-06:
I have change the coding to this [...
FirebaseJson json;
FirebaseJsonArray jsonArr;

jsonArr.add(measure);
json.clear().add("data", jsonArr).add("timestamp", 0);
...]
The error is 
[
sketch\cardio_sender.ino.cpp.o:(.literal._Z4loopv+0x3c): undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

sketch\cardio_sender.ino.cpp.o: In function `loop()':

C:\Users\USER\Downloads\cardio_sender/cardio_sender.ino:60: undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

collect2.exe: error: ld returned 1 exit status
]

mobizt, 2020-06-06:
Please copy all error messages and post them again.

TeohWeiWei, 2020-06-06:
This is the error:
sketch\cardio_sender.ino.cpp.o:(.literal._Z4loopv+0x3c): undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

sketch\cardio_sender.ino.cpp.o: In function `loop()':

C:\Users\USER\Downloads\cardio_sender/cardio_sender.ino:60: undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "SD.h"
 Used: C:\Users\USER\Documents\Arduino\hardware\espressif\esp32\libraries\SD
 Not used: C:\Program
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\USER\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program
exit status 1
Error compiling for board ESP32 Dev Module.

mobizt, 2020-06-06:
Look like it's not completed error message.

Your installation for ESP32 Arduino Core SDK does not complete or use the outdated installation method.

Remove this folder  **C:\Users\USER\Documents\Arduino\hardware\espressif\esp32**

TeohWeiWei, 2020-06-06:
Error: 
Arduino: 1.8.11 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

sketch\cardio_sender.ino.cpp.o:(.literal._Z4loopv+0x3c): undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

sketch\cardio_sender.ino.cpp.o: In function `loop()':

C:\Users\USER\Downloads\cardio_sender/cardio_sender.ino:60: undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\USER\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program
Multiple libraries were found for "SD.h"
 Used: C:\Users\USER\Documents\Arduino\hardware\espressif\esp32\libraries\SD
 Not used: C:\Program
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mobizt, 2020-06-06:
You can remove folder **C:\Users\USER\Documents\Arduino\hardware** as it was redundant and not necessary.

Follow the install instruction [here](https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/).

Remove Firebase library and install it again from Library Manager.

closed 2020-06-06 by mobizt

TeohWeiWei, 2020-06-06:
Hi, I have reinstalled the ESP32 and the firebase library by following your instruction, but still have the error

This attachment is my arduino coding
[ECG_ArduinoCoding.txt](https://github.com/mobizt/Firebase-ESP32/files/4739838/ECG_ArduinoCoding.txt)

Error:
Arduino: 1.8.11 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

sketch\cardio_sender.ino.cpp.o:(.literal._Z4loopv+0x40): undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

sketch\cardio_sender.ino.cpp.o: In function `loop()':

C:\Users\USER\Downloads\cardio_sender/cardio_sender.ino:60: undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "SD.h"
 Used: C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD
 Not used: C:\Program
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mobizt, 2020-06-06:
The problems still exist because you install some third-party libraries that redundant and conflicts with the core library used in the Firebase library.

I don't know which library causes the problem unless SD.h (found in multiple libraries) at this time.

For sure, you need to remove all libraries in the libraries folder
**C:\Users\USER\Documents\Arduino\libraries** 

Then reinstall Firebase library again.

TeohWeiWei, 2020-06-06:
I have remove all of the libraries in the libraries folder
C:\Users\USER\Documents\Arduino\libraries and reinstall the firebase library but still got the error.
Error:  
Arduino: 1.8.11 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

sketch\cardio_sender.ino.cpp.o:(.literal._Z4loopv+0x40): undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

sketch\cardio_sender.ino.cpp.o: In function `loop()':

C:\Users\USER\Downloads\cardio_sender/cardio_sender.ino:60: undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "SD.h"
 Used: C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD
 Not used: C:\Program
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

I try to move the firebase library to C:\Program Files (x86)>Arduino> libraries and still have the error too. There are a lot of the libraries in the C:\Program Files (x86)>Arduino> libraries, Do I need to delete all of them as well? I only deleted all libraries which in C:\Users\USER\Documents\Arduino\libraries

mobizt, 2020-06-06:
The user library is at C:\Users\USER\Documents\Arduino\libraries which may cause conflicts from bad design libraries.

The Arduino default core library in  C:\Program Files (x86)\Arduino\libraries is ok and don't touch it.

Your error is really strange and missing the useful information for me to investigation

For example

**Used: C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
Not used: C:\Program** <---- it should show the full file path  instead of **C:\Program**

Your computer user name should not contains space and contains only the Arabic alphabet. 

You need to test your Arduino IDE installation, the sketch folder (libraries folder) must not lock by another program for reading and writing e.g Microsoft OneDrive.

If you can install the third party library and successfully compiled, you can install and compile Firebase library too.

TeohWeiWei, 2020-06-06:
I have used another laptop (my SISTER's laptop) to install a new Arduino application and redo all those step. 

In the error has shown the useful information that you have mentioned about.
Next, I have solve this error [Multiple libraries were found for "WiFi.h"] by adding the library from the Sketch>Include Library> Contributed Library>WiFi

Library added to solve error: [Multiple libraries were found for "WiFi.h"]:
#include <ETH.h>
#include <WiFi.h>
#include <WiFiAP.h>
#include <WiFiClient.h>
#include <WiFiGeneric.h>
#include <WiFiMulti.h>
#include <WiFiScan.h>
#include <WiFiServer.h>
#include <WiFiSTA.h>
#include <WiFiType.h>
#include <WiFiUdp.h>

This is the Error: 
Arduino: 1.8.12 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

sketch\cardio_sender.ino.cpp.o:(.literal._Z4loopv+0x3c): undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

sketch\cardio_sender.ino.cpp.o: In function `loop()':

C:\Users\Teoh WeiTing\Downloads\cardio_sender/cardio_sender.ino:74: undefined reference to `FirebaseJsonArray& FirebaseJsonArray::add<unsigned char>(unsigned char)'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "SD.h"
 Used: C:\Users\Teoh WeiTing\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mobizt, 2020-06-06:
I've checked your code and seen that you pass the uint8_t variable to the FirebaseJsonArray **add** function, **jsonArr.add(measure)**, which does not match the function templates.

For FirebaseJson and FirebaseJsonArray, the integer number, you should use **int** (uint8_t, uint16_t, uint32_t, and long data types are not supported) and for the floating-point number, you should use **double** (float data type is not supported).

In your sketch, you have to change from **uint8_t measure = 0;** to **int measure = 0;**.

TeohWeiWei, 2020-06-07:
For the conclusion, I have only changed these few codes. Thank god it's able to compile successfully. 
Thanks Mobizt for the fast reply!!! I really appreciated to it T_T 
[
uint8_t previous_measure = 0; 
uint8_t measure = 0; 
jsonArr.addInt(measure);
json.clear().addArray("data", &jsonArr).addInt("timestamp", 0); 
]
to
[
int previous_measure = 0;
int measure = 0;
jsonArr.add(measure);
json.clear().add("data", jsonArr).add("timestamp", 0);
]