mobizt/Firebase-ESP32 #76: Trying to connect a TT goT call (esp32wrover ) via gprs . internet coonected but fire base eror "connection lost ".  where i am doing wrong?

biswajitjei02, 2020-06-07:
// TTGO T-Call pin definitions
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22
// Other define
#define BLYNK_PRINT Serial    
#define TINY_GSM_MODEM_SIM800

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define Serial Serial
// Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

// Include
#include <TinyGsmClient.h>
#include <BlynkSimpleTinyGSM.h>
#include <Wire.h>
#include "utilities.h"

// Vars
const char apn[]  = "portalnmms";
const char user[] = "";
const char pass[] = "";
// Go to the Project Settings (nut icon).
const char auth[] = "XXXXXXXtkjKY5JsANxUF2WmOXKP_";

TinyGsm modem(SerialAT);
BlynkTimer timer;

void checkGPRS()
{
  Serial.println("GPRS check running");

  if (!modem.isGprsConnected()) {
    if (!modem.gprsConnect(apn, user, pass)) {
      delay(10000);
      Serial.println("GPRS not running");
      return;
    }
  }
}

#include <FirebaseESP32.h>
#define FIREBASE_HOST "XXXXXX.firebaseapp.com"
#define FIREBASE_AUTH "XXXXXXXXXXXXXXXXXXfhpFdfG1w7HsNzsdf"

 FirebaseData firebaseData;

int val;
void setup()
{

  
  // Set console baud rate
  Serial.begin(115200);
  

  // Keep power when running from battery
  Wire.begin(I2C_SDA, I2C_SCL);
  bool   isOk = setPowerBoostKeepOn(1);
  Serial.println(String("IP5306 KeepOn ") + (isOk ? "OK" : "FAIL"));

  // Set-up modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);

  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);
  digitalWrite(MODEM_POWER_ON, HIGH);

  // Set GSM module baud rate and UART pins
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
  delay(3000);

  

  
  // For GPRS reconnect
  timer.setInterval(10000L, checkGPRS);
 
  Firebase.begin("XXXXXXXX..firebaseio.com","XXXXXXXXXXXXXXXXXFdfG1w7HsNzsdf");
}

void loop()
{
   

  Firebase.getString(firebaseData, "newvalue");
   Serial.println(firebaseData.stringData());
  Serial.println(firebaseData.errorReason());
   
delay(1000);
  timer.run();
    

}

biswajitjei02, 2020-06-07:
utilities,h

#define IP5306_ADDR 0x75
#define IP5306_REG_SYS_CTL0 0x00

bool setPowerBoostKeepOn(int en)
{
Wire.beginTransmission(IP5306_ADDR);
Wire.write(IP5306_REG_SYS_CTL0);
if (en) {
Wire.write(0x37); // Set bit1: 1 enable 0 disable boost keep on
} else {
Wire.write(0x35); // 0x37 is default reg value
}
return Wire.endTransmission() == 0;
}

mobizt, 2020-06-07:
The library supports only WiFi, any GPRS bridge was not supported.

closed 2020-06-07 by mobizt