mobizt/Firebase-ESP32 #78: how to read a Jason?

biswajitjei02, 2020-06-11:
[image]
here  want a Jason like this 
{newvalue: 32, value: 1,value1: "no"}

mobizt, 2020-06-11:
If you want to get the data at the root of your database, use **Firebase.getJSON** with the path parameter "/"

```C++
  Firebase.getJSON(firebaseData, "/");

  FirebaseJson &json = firebaseData.jsonObject();

  FirebaseJsonData parseRsult;

  json.get(parseRsult, "/newvalue");

  if (parseRsult.success)
  {
    Serial.println(parseRsult.intValue);
  }

  json.get(parseRsult, "/value");

  if (parseRsult.success)
  {
    Serial.println(parseRsult.intValue);
  }

  json.get(parseRsult, "/value1");

  if (parseRsult.success)
  {
    Serial.println(parseRsult.stringValue);
  }
```

**For more usage of the FirebaseJson object**

 https://github.com/mobizt/Firebase-ESP32#parse-create-and-edit-json-objects
https://github.com/mobizt/Firebase-ESP32/blob/master/examples/jsonObject/jsonObject.ino
https://github.com/mobizt/Firebase-ESP32/tree/master/examples/FirebaseJson/examples

biswajitjei02, 2020-06-11:
i want to store as a single Json a ,how it can be done ?

mobizt, 2020-06-11:
I don't understand what you mean.

mobizt, 2020-06-11:
Please read the Readme doc or start with the examples.

closed 2020-06-11 by mobizt

biswajitjei02, 2020-06-11:
this type of out put i want 
//Declare FirebaseJson object (global or local)
FirebaseJson json;

//Add name with value Living Room to JSON object
json.add("name", "Living Room");

//Add temp1 with value 120 and temp1 with 40 to JSON object
//Note: temp2 is not the child of temp1 as in previous version.
json.add("temp1", 120).add("temp2", 40);

//Add nested child contents directly
json.set("unit/temp1", "Farenheit");
json.set("unit/temp2", "Celcius");

//To print out as prettify string
String jsonStr;
json.toString(jsonStr, true);
Serial.println(jsonStr);

/*
This is the result of the above code

{
    "name": "Living Room",
    "temp1": 120,
    "temp2": 40,
    "unit": {
        "temp1": "Farenheit",
        "temp2": "Celcius"
    }
}