mobizt/Firebase-ESP32 #79: HELP

am-canes, 2020-06-12:
Is there any way to know if firebase is connected? I would like to send set and get just if firebase is connected. 
And is there any way to know the result of firebase.begin? Like an error if firebase host or authentication are incorrect.

Thanks.

mobizt, 2020-06-12:
The function Firebase.begin is just save the authentication credentials.

All examples provided showed the error debugging in every Firebase calls.

https://github.com/mobizt/Firebase-ESP32/tree/master/examples

closed 2020-06-12 by mobizt

am-canes, 2020-06-12:
Ok, thanks @mobizt.
So I just need to call .begin when I start the code, it will have the credentials saved even if the wifi is disconnected and connected again later?

mobizt, 2020-06-13:
As you seen, the function **Firebase.begin** takes the Firebase authentication credential, no WiFi begin operation involved unless you set the [reconnect](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Basic/Basic.ino#L52) function which will call reconnect function of WiFiClient core library periodically.

am-canes, 2020-06-14:
Yes, I understood it. I am handling with wifi reconnect in another part of my code. Thanks