mobizt/Firebase-ESP32 #80: HELP - Got zero value at firebase and even serial monitor

yys100, 2020-06-20:
Hi, @mobizt. I have utilize the firebase-esp32 library (working on dfrobot heart rate sensor and nodemcu esp 32). Unfortunately when I try to send the sensor data using wifi to firebase, I got Zero/0 value on both firebase and serial monitor on arduino ide. I have been searching for solution but still not able to tackle this issue. I am hoping you spare some time to help me out with this. Any help/guidance will be greatly appreciated. Thanks in advance.

Partial code snippet of my program on arduino ide:
```
void loop() {
  uint8_t rateValue;
  heartrate.getValue(heartratePin); ///< A1 foot sampled values
  rateValue = heartrate.getRate(); ///< Get heart rate value 
  /*if(rateValue)  {
    Serial.println(rateValue);
  }*/

   if(Firebase.setInt(firebaseData,"HeartBeat/BPM_data", rateValue)){
    if (firebaseData.dataType() == "int") {
      Serial.println(firebaseData.intData());
      //Serial.println(rateValue);
      Serial.println("testing");
    }
     else {
    Serial.println(firebaseData.errorReason());
  }
  }
  delay(20);
}
```

Hope to hear from you soon and thanks!

Regards
yys100

mobizt, 2020-06-21:
You should use **int** (32-bit) or **uint16_t** to get the value (16-bit) from sensor instead of **uint8_t** (8-bit).

Here is the [example](https://github.com/DFRobot/DFRobot_Heartrate/blob/master/examples/DFRobot_Heartrate_Analog_Mode/DFRobot_Heartrate_Analog_Mode.ino) from DFRobot.

And the GPIO you assigned for **heartratePin** should not one of the ADC1 channels because ADC1 can't use (some GPIOs are not operate as ADC) when WiFi is operating.

Here are the pins you can't use as analog input during WiFi enabled, GPIO 36, GPIO 39, GPIO 34, GPIO 35, GPIO 32 and GPIO 33.

You can use ADC2 channels instead i.e. GPIO 25, GPIO 26, GPIO 27, GPIO 14, GPIO 12, GPIO 13, GPIO 4, GPIO 2, GPIO 15.

closed 2020-06-21 by mobizt

yys100, 2020-06-21:
Hi, thanks for getting back to me. I am sorry as I forgot to mention that I am using the digital mode for the Dfrobot heart rate sensor with nodemcu esp32. [example_digitalMode](https://github.com/DFRobot/DFRobot_Heartrate/blob/master/examples/DFRobot_Heartrate_Digital_Mode/DFRobot_Heartrate_Digital_Mode.ino) 

For the advise you have stated, is it still applicable ? The example is using uint8_t (8-bit) and I am not sure do I need to change it.  For the input pin as well, this I assume is different from if I use it as digital mode?

I hope you help me clarify my issue, thanks!

Regards,
yys100

mobizt, 2020-06-21:
In the sensor library, it reads the analog value from GPIO pin (ADC) no matter you config it in digital or analog mode.

You never provide the information about your declaration of **heartratePin**.

Read  my recommendation in my above post again for ADC GPIO that can use.

It is safe to store the value in **integer** as my library accept only integer (uint8, uint16, uint32 are not support).

This is not the forum for question that not related to my library.

You should open the issue at DFRobot github.

yys100, 2020-06-21:
Hi, really appreciate that you could get back to me in such a short time. I have posted my issue on the dfrobot forum but I still not getting any reply yet. Could you pls advise what I can do with uint8_t ? Is it possible to convert it to Int, is there an example to show? 

I think it would be better if I show you my program for better clarification, hope you will understand.
```
**#define heartratePin 25 //change as you have advised**
#include "DFRobot_Heartrate.h"
#include <WiFi.h>  //1. Include WiFi.h
#include "FirebaseESP32.h"  //2. Include Firebase ESP32 library (this library)

// Set these to run example.
#define FIREBASE_HOST "test.firebaseio.com"  
#define FIREBASE_AUTH "hW8qlM48dGkxxxxxxxx1Pq5UD21YgPq2E0c"
#define WIFI_SSID "Gummya"
#define WIFI_PASSWORD "testtest1"

DFRobot_Heartrate heartrate(DIGITAL_MODE); ///< ANALOG_MODE or DIGITAL_MODE
FirebaseData firebaseData;  //3. Declare the Firebase Data object in the global scope

void setup() {
  Serial.begin(115200);
  initWifi();
}

void loop() {
  uint8_t rateValue;
  heartrate.getValue(heartratePin); ///< A1 foot sampled values **//Calling mutiple times? , is this right**
  rateValue = heartrate.getRate(); ///< Get heart rate value
   if(Firebase.setInt(firebaseData,"HeartBeat/BPM_data", rateValue)){
    if (firebaseData.dataType() == "int") {
      Serial.println(firebaseData.intData());
      Serial.println(rateValue);
      Serial.println("testing");
    }
     else {
    Serial.println(firebaseData.errorReason());
  }
  }

  **//delay(20); //delay removed**
}

 void initWifi(){
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  
  //Set database read timeout to 1 minute (max 15 minutes)
  Firebase.setReadTimeout(firebaseData, 1000 * 60);
  //tiny, small, medium, large and unlimited.
  //Size and its write timeout e.g. tiny (1s), small (10s), medium (30s) and large (60s).
  Firebase.setwriteSizeLimit(firebaseData, "tiny");
 }
```

Thanks!
Regards
yys100

mobizt, 2020-06-21:
You should test with the example from DFRobot first without Firebase and WiFi and assigned GPIO 25 for sensor pin to check your wiring and sensor function are ok.

If the above test does not work, you should check the sensor wiring and power supply.

Your sensor may defect or the supply voltage issue or open circuit in wiring.

If above test works, then try the following code.

```C++
#define heartratePin 25 //change as you have advised**
#include <DFRobot_Heartrate.h>
#include <WiFi.h>          //1. Include WiFi.h
#include <FirebaseESP32.h> //2. Include Firebase ESP32 library (this library)

// Set these to run example.
#define FIREBASE_HOST "test.firebaseio.com"
#define FIREBASE_AUTH "hW8qlM48dGkxxxxxxxx1Pq5UD21YgPq2E0c"
#define WIFI_SSID "Gummya"
#define WIFI_PASSWORD "testtest1"

DFRobot_Heartrate heartrate(DIGITAL_MODE); ///< ANALOG_MODE or DIGITAL_MODE
FirebaseData firebaseData;                 //3. Declare the Firebase Data object in the global scope

int rateValue;

static int taskCore = 1;

void initWifi();

void sensorLoopTask(void *pvParameters)
{
  //run forever
  while (true)
  {
    heartrate.getValue(heartratePin); ///< A1 foot sampled values **//Calling mutiple times? , is this right**
    rateValue = heartrate.getRate();  ///< Get heart rate value
    delay(20);
  }
}

void setup()
{
  Serial.begin(115200);
  initWifi();
  
  //run sensor reading task forever in an another RTOS task
  xTaskCreatePinnedToCore(
      sensorLoopTask, /* Function to implement the task */
      "sensorTask",   /* Name of the task */
      10000,          /* Stack size in words */
      NULL,           /* Task input parameter */
      0,              /* Priority of the task */
      NULL,           /* Task handle. */
      taskCore);      /* CPU Core where the task should run */
}

void loop()
{

  if (Firebase.setInt(firebaseData, "HeartBeat/BPM_data", rateValue))
  {
    if (firebaseData.dataType() == "int")
    {
      Serial.println(firebaseData.intData());
      Serial.println(rateValue);
      Serial.println("testing");
    }
    else
    {
      Serial.println(firebaseData.errorReason());
    }
  }

  delay(2000);

  //delay(20); //delay removed**
}

void initWifi()
{
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  //Set database read timeout to 1 minute (max 15 minutes)
  Firebase.setReadTimeout(firebaseData, 1000 * 60);
  //tiny, small, medium, large and unlimited.
  //Size and its write timeout e.g. tiny (1s), small (10s), medium (30s) and large (60s).
  Firebase.setwriteSizeLimit(firebaseData, "tiny");
}

```

mobizt, 2020-06-21:
You should change the database secret for FIREBASE_AUTH for privacy and security because you post it to public here.

yys100, 2020-06-21:
> You should test with the example from DFRobot first without Firebase and WiFi and assigned GPIO 25 for sensor pin to check your wiring and sensor function are ok.
>
> If the above test does not work, [...]

Hi, @mobizt. Thanks for the code provided, I have tested/run it. However, I still got Zero/0 error and there's no error/msg indicates what is the actual reason the values keep showing 0. 

@mobizt, do you know what the reason for this? besides, I will take note and change the database secret. Thanks!

mobizt, 2020-06-21:
example from DFRobot is ok?

mobizt, 2020-06-21:
I recommend you to change database secret because now everyone that read your post know this information and can edit or modify your database data.

yys100, 2020-06-21:
> example from DFRobot is ok?

Yes it works fine (without the usage of firebase and wifi connection)
Although the sensor value take some time to be read on the serial monitor

mobizt, 2020-06-21:
you may need to check the rateValue in loop before send to firebase.

```
if (rateValue > 0)
{

//send data to firebase

}
```

mobizt, 2020-06-21:
OK my information of ADC is wrong. I also correct my above post.

You need to use ADC1 instead of ADC2.

You can use ADC1 i.e. GPIO 36, GPIO 39, GPIO 34, GPIO 35, GPIO 32 and GPIO 33 for analog input when WiFi enable.

yys100, 2020-06-21:
> you may need to check the rateValue in loop before send to firebase.
> 
> ```
> if (rateValue > 0)
> {
> 
> //send data to firebase
> 
> }
> ```

hi I have tested it, it just prints "NAH". What does it means tho? I failed to send the data to firebase? or it might be the sensor problem?  I have changed the GPIO 25 --> GPIO33

```
void loop()
{
  if (rateValue > 0)
  {
      //send data to firebase
      Firebase.setInt(firebaseData, "HeartBeat/data", rateValue);
      Serial.println(rateValue);
  }
  else
  {
      Serial.println("Nah");
  }
  delay(2000);
}
```

Regards
yys100

mobizt, 2020-06-21:
It means rateValue value is zero.

yys100, 2020-06-21:
> It means rateValue value is zero.

Sorry, I don't seem to get it. The value of heart beat are able to obtain while i try it with the dfrobot example. I just utilized it with firebase and wifi and it turn out the value to be zero? hmm how does it work ? Hope you could help to clear my doubts. Thanks 

Regards 
yys100

mobizt, 2020-06-21:
I can't provide further help because I don't have this kind of sensor to test.

You need to test you own with WiFi client or http client examples (examples are available in Arduino IDE) while reading the sensor value.

If it does not work you may try other GPIO pins.

mobizt, 2020-06-22:
OK I've found that you never assign the ADC channels to pin. You must call this in setup.

**adcAttachPin(heartratePin);**