mobizt/Firebase-ESP32 #81: HELP - Why my issue deleted? #80

yys100, 2020-06-22:
Hi, @mobizt. Why is my issue being deleted, I received the email you commented on my issue but when I clicked on it, it shown that the issue was deleted. 

I just want to refer to the advise/suggestion you have provided. Therefore, the conversation we had is really important. Hope you could undo the deleted issue. #80 

thanks

mobizt, 2020-06-22:
Yes I delete because it's not **library issue** as I inform and this will delete too.
And hope you got my last reply in email and please follow it, your problem can solve now.

closed 2020-06-22 by mobizt

yys100, 2020-06-22:
`adcAttachPin(heartratePin);`

how does this work tho? I kinda confused tho 
do I type like this adcAttachPin(33);

mobizt, 2020-06-22:
Yes, you can and you should use 33 in the DFRobot sensor function too

mobizt, 2020-06-22:
ESP32 need to config ADC like this before use.

yys100, 2020-06-22:
Hi, I play around with code again but this time it just keep showing "wait for valid data" at the serial monitor and sometimes print zero(very rare) but shows diff value at the database. I assume the sensor is not stable ?? is it right? 

Thanks for all the help after all, really appreciated!