mobizt/Firebase-ESP32 #82: is Firebase Push message is available?

happynet95, 2020-07-06:
Hello. I am looking for a library that makes ESP32 receive push messages sent from web.

Your library seems to contain functions about sending messages from ESP to server, but I'm not sure whether it can receive messages.

Can you give me the answer?

thank you.

mobizt, 2020-07-06:
If you mean Firebase Cloud Messaging, the library does not provide the receiving of push message which it's not available from the REST APIs used in the library and may be available on other platform e.g. web, Android etc.

mobizt, 2020-07-06:
You can create  your own TCP socket server in PHP, C#, or C++ to connect with client (esp32). Which you can push the message in real-time between server and client.

closed 2020-07-06 by mobizt

happynet95, 2020-07-06:
Thank you for the answer! :)

yatian-liu, 2020-10-20:
> You can create your own TCP socket server in PHP, C#, or C++ to connect with client (esp32). Which you can push the message in real-time between server and client.

Excuse me, it is possible to integrate the TCP socket server with Firebase? For example, the server can first receive a push message from Firebase indicating data updates and then fetch the new data and send it to ESP32 via TCP?

mobizt, 2020-10-20:
@yatian-liu 

I never try it and not sure.  Let try it.

You can install Firebase Admin SDK to your server that run Node.js as TCP socket server and use the reference of the database that created to  send the data to ESP32.

For Firebase Admin SDK using Node.js
https://firebase.google.com/docs/database/admin/start#node.js

For TCP socket server (many via google search)
https://www.yld.io/blog/building-a-tcp-service-using-node-js/